PX4/PX4-Autopilot #10146: EKF2 not stable and restarts after amount in time when using it with vision_pose estimation

kawasakiDaichi, 2018-08-04:
**Describe the bug**
A clear and concise description of the bug.
I have quadcopter with down-looking camera(oCam) and odroid xu4 and ros installed I got everything works after fixing interference problems with UART TELEM2 line but when i feed `svo/pose` to vi_ekf to fuse it with imu measurements  and publish output to `mavros/vision_pose/pose` in 80Hz/10Hz(tested both). i got instability in the `mavros/local_position/pose` and ekf2 keep restarting after little random time (about 2 - 10 seconds)

the hardware i use Pixhawk 1 with 2 MB Flash and Odroid XU4 connected with USB-TTL to TELEM2 and Companion Link at 921600 baudrate and use compiled stock PX4  v1.8.0 just added  this fix [https://github.com/PX4/ecl/commit/e1661a92c379375cbcdbb50192dd2330b8d57d24](url) for vision initialization and compiled with px4fmu-v3_default configs

Notice that ekf2 after restarting is follows the vision estimation and then it drifts far from it without too much too much change in vision pose (quadcopter it stationary) 

My parameters:
EKF2_AID_MASK: 8 (just Vision pose)
EKF2_HGT_MODE: Vision
EKF2_DECL_TYPE: 0
EKF2_GPS_CHECK: 0

the echo of `mavros/vision_pose/pose ` at 10 Hz publishing rate:
```
header: 
  seq: 1154
  stamp: 
    secs: 1455212670
    nsecs: 346535442
  frame_id: "map"
pose: 
  position: 
    x: 0.340464502573
    y: -0.621057868004
    z: -2.58103728294
  orientation: 
    x: -0.00563598191366
    y: -0.00214419327676
    z: 0.254346847534
    w: 0.967094242573
---
header: 
  seq: 1155
  stamp: 
    secs: 1455212670
    nsecs: 447407599
  frame_id: "map"
pose: 
  position: 
    x: 0.341008067131
    y: -0.621058881283
    z: -2.58947992325
  orientation: 
    x: -0.00537286885083
    y: -0.00225412286818
    z: 0.254378139973
    w: 0.967087268829
---
header: 
  seq: 1156
  stamp: 
    secs: 1455212670
    nsecs: 546944543
  frame_id: "map"
pose: 
  position: 
    x: 0.340965896845
    y: -0.620927810669
    z: -2.59085273743
  orientation: 
    x: -0.00539355212823
    y: -0.00249927560799
    z: 0.25435769558
    w: 0.967091917992
---
header: 
  seq: 1157
  stamp: 
[... 419 lines cut ...]
<node name="vi_fusor" pkg="vi_ekf" type="vi_ekf" output="screen" launch-prefix="taskset -c 2">
	<remap from="teensy/imu" to="mavros/imu/data" />
	<remap from="ekf/pose" to="mavros/vision_pose/pose" />
</node>

    <node name="ocam_ros" pkg="ocam_ros" type="ocam_ros_node" output="screen" >
      <param name="image_topic" value="/camera/image_raw" />
      <param name="undistort" value="false" />
      <param name="auto_exposure" value="false" />
      <param name="exposure" value="80" />
      <param name="brightness" value="200" />
    </node>
</launch>
```
2- initialize svo slam by moving quadcopter on planar surface
3- EKF2 take measurement after start and start to drift after a while it restarts and follow the vision pose and then drifts and so on (without moving quadcopter)

**Expected behavior**
I think of the EKF2 tunings or hardware problem cause i tested with LPE in v1.7.0 and gives me every 2 seconds : 
```
 [ INFO] [1469985463.843807007]: FCU: [lpe] vision position init: 4.01, -4.14, -0.98 m
 [ INFO] [1469985465.355433811]:  FCU: [lpe] vision position timeout
```

and sorry for my bad english

EdXian, 2018-08-06:
Hi @kawasakiDaichi ,
Did you solve the problem?

kawasakiDaichi, 2018-08-06:
Nope!

tiralonghipol, 2018-09-07:
Hi. Any news on this ?

Doodleshr, 2018-09-12:
same problem here, any help?

closed 2019-02-18 by stale[bot]

lgh5054, 2019-07-10:
@Doodle1106 Have you solved this problem?

julianoes, 2019-07-15:
So it looks like the issue is with EKF2 and LPE, correct? If it persists and you have not found a solution, please re-open.