PX4/PX4-Autopilot #10162: optical flow position mode - "no local position" error

ambikaverma, 2018-08-05:
Hi

Im trying to test optical flow on intel aero. My setup is as follows -

px4 - ver 1.8.0
bios - latest yocto image.
indoors, when the uav is stationary.

I start optical flow manually using ./aero-optical-flow command to see if the features/texture is good enough. Looks like its good enough since I don't get any desired number of feature prompts. Also the quality is on 180-200 range.

And on arming the flight mode does change to position hold.
But, instantly I get "no local position" error. I can share screenshots of above if required.

Any suggestions on the cause for this?
Thanks!

ambikaverma, 2018-08-06:
Any fixes of above issue?

ambikaverma, 2018-08-13:
as per here- 
https://github.com/intel-aero/meta-intel-aero/issues/376

I would need to debug px4. How do I go about doing that? 
I tested altitude hold and the flight went fine, but optical flow position hold is still going into failsafe of no local position.

ChristophTobler, 2018-08-13:
Can you please provide a log?

closed 2019-02-18 by stale[bot]