PX4/PX4-Autopilot #10163: In EKF2 range finder is not getting data from distance sensor data

kyuhyong, 2018-08-05:
**Describe the bug**
A clear and concise description of the bug.
Distance sensor is coming from sonar and is updating correctly as shown in QGC analyzer.
[image]

In EKF2 I added a debug message in range finder section to see if this data is also getting into ekf2 
estimator. 
When checked this message in the analyzer, the value is not updating as expectecd.
[image]

Is range finder in EKF2 is different than distance sensor?
Below is the code I modified in /src/modules/ekf2/ekf2_main.cpp

```
		if (_range_finder_sub_index >= 0) {
			bool range_finder_updated = false;
			orb_check(_range_finder_subs[_range_finder_sub_index], &range_finder_updated);
			if (range_finder_updated) {
				distance_sensor_s range_finder;
				if (orb_copy(ORB_ID(distance_sensor), _range_finder_subs[_range_finder_sub_index], &range_finder) == PX4_OK) {
					// check distance sensor data quality
					// TODO - move this check inside the ecl library
					if (range_finder.signal_quality == 0) {
						// use rng_gnd_clearance if on ground
						if (_ekf.get_in_air_status()) {
							range_finder_updated = false;
						} else {
							range_finder.current_distance = _rng_gnd_clearance.get();
						}
					}
					if (range_finder_updated) { _ekf.setRangeData(range_finder.timestamp, range_finder.current_distance); }
					// Save sensor limits reported by the rangefinder
					_ekf.set_rangefinder_limits(range_finder.min_distance, range_finder.max_distance);
					ekf2_timestamps.distance_sensor_timestamp_rel = (int16_t)((int64_t)range_finder.timestamp / 100 -
							(int64_t)ekf2_timestamps.timestamp / 100);
					my_dbg.value = range_finder.current_distance;
					orb_publish(ORB_ID(debug_key_value), pub_dbg, &my_dbg);
				}
			}
		} else {
			_range_finder_sub_index = getRangeSubIndex(_range_finder_subs);
		}
```

dakejahl, 2018-08-05:
Make sure you set the `signal_quality = 1` variable in the `distance_sensor_report_s`. I'm also aware of a breaking issue (at least for us) with the way the ekf2 is fusion range finder data and working on a fix. Let me know if this gets it working or if you run into further issues. I am curious if the bug we are currently experiencing with range finder fusion is affecting anyone else.

LorenzMeier, 2018-08-05:
@priseborough Could you please have a look? Thanks!

dakejahl, 2018-08-05:
@kyuhyong also make sure you are setting the parameter EKF2_RNG_AID

kyuhyong, 2018-08-06:
@dakejahl 
I did check EKF2_RNG_AID is set tto 1.
I found the signal_quality = 0 all the time so I manually set it to 1.
Now the range_finder.current_distance is updating but not sure if ekf2 is using this data or not.
Still cannot get correct altitude hold with distance sensor.
Here is my log. https://logs.px4.io/plot_app?log=97a597d4-ef14-47b5-ad11-f49e5a86bac1

kyuhyong, 2018-08-06:
Now I played altitude hold without px4flow and getting some weird behavior. 
It looks like ekf2 is using distance data however jumps erratically. At some point it jumps to the roof and I had to throttle down to idle. 
[image]

https://logs.px4.io/plot_app?log=3ec7d0d0-5df3-4157-b5a7-d3dd4e70d307

dakejahl, 2018-08-06:
You need to do some filtering on that range finder data at the driver level. Notice how the bad measurements show up in the **Local Position Z** estimate that you shared above.
[image]

kyuhyong, 2018-08-07:
@dakejahl Yes I noticed that too. I thought EKF2 will handle those noise by updating covariance but obviously not. Is there any references I can have a look to apply filtering in driver level? 
Otherwise I have to make any averaging or complementary filter?
Thanks,

mhkabir, 2018-08-07:
Please use MPC_ALT_MODE=Terrain Following, not Range Aid.

kyuhyong, 2018-08-07:
I already set MPC_ALT_MODE=Terrain Following. 
You mean disable Range aid?

mhkabir, 2018-08-07:
Yes, you don't require range aid if you're in terrain following already.

kyuhyong, 2018-08-07:
Here is another log. 
https://logs.px4.io/plot_app?log=701dd9eb-8890-415b-8776-ac60ccec1d7c
Still not following the range sensor and at one point it shoots to the roof.
[image]

dakejahl, 2018-08-07:
@kyuhyong You'll probably need to do more than just a moving average filter. I'd start with trying to improve the quality of the data your hardware is generating. Mess with your ASICs settings. By looking at your raw data, the thresholds from 0 to 1m are set too low, thats why you are getting so much noise in that region. After that, you'll want to throw away data points that are blatantly wrong i.e:
`if(current_meas > prev_meas + meas_noise || current_meas < prev_meas - meas_noise)`
From there you can drop on a filter of your choosing and get much better results. It's impossible for a filter to provide a good estimate of state if you are constantly feeding your filter bad data.

Chraebi, 2018-11-18:
Dear community,
I'm not quite sure, why you don't need EKF2_RNG_AID in the MPC_ALT_MODE. Since the EKF2 only influences the estimator of the states which then are used in the MP-Controller. So I thought it is benfeficial, if the controller gets more accurate data from the estimator.
Kind regards and thanks for the help

priseborough, 2018-11-18:
EKF2_RNG_AID is only suitable for use over either flat surfaces or for vertical takeoff and landings where horizontal flight is performed above the range finder switchover height.

Where terrain is varying and the vehicle is flying horizontally, using range finder in the EKF is not advised.

priseborough, 2018-11-19:
You can give it a try. Note that when the EKF repeatedly switches between Baro and range finder due to horizontal manoeuvring, the combination of tilt, attitude estimation error range finder alignment, static pressure field etc can cause 'ratcheting' of the height offset, so depending on your application it could give a worse result.

Chraebi, 2018-11-19:
Thank you very much for your valuable input!
Kind regards

closed 2019-03-03 by stale[bot]