PX4/PX4-Autopilot #10164: Ensure that environment variables are disabled in NuttX

LorenzMeier, 2018-08-05:
We are not using environment variables in NuttX. They are consuming RAM though. We should make sure this is disabled in NuttX. @davids5 Could you have a quick pass there?

davids5, 2018-09-27:
Two options:

option 1) upstreamable is to add export  to nsh.
option 2) is not upstream-abe to add an  #if def CONFIG_ENVIRON_INHERITANCE and loose env_dup duplication.

CONFIG_ENVIRON_INHERITANCE=y
```
1591000	   4269	  32968	1628237	 18d84d	build/nuttx_px4fmu-v5_default/nuttx_px4fmu-v5_default.elf

nsh> free
             total       used       free    largest
Mem:        486208     182096     304112     300672
```
but the gain is only small  presumably based on fragmentation

with CONFIG_ENVIRON_INHERITANCE = no
```
1590896	   4269	  32968	1628133	 18d7e5	build/nuttx_px4fmu-v5_default/nuttx_px4fmu-v5_default.elf
nsh> free
             total       used       free    largest
Mem:        486208     182080     304128     301216
```
Do you think it is worth it?

dagar, 2018-09-27:
Any options that would eliminate the need for this? https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d/rcS#L587-L619

davids5, 2018-09-27:
Neither will not do that. :(  Nsh is still running