PX4/PX4-Autopilot #10167: Hitting invisible barrier using range finder

dakejahl, 2018-08-05:
**Describe the bug**
Current `master` flying with EKF2_RNG_AID on MC airframe. I am hitting an invisible wall when trying to throttle up. This occurs with using `signal_quality = 0` as well as with not publishing data I know is bad. If I turn off the range finder, I do not see this behavior. A lot has changed with the range finder logic in the `ecl` over the last month and a half, and I suspect this bug crept in with those changes. Overall, the range aiding feature has never really worked well for us. I'd like to see some form of real fusion between baro/range finder as opposed to our current solution which results in toggling between the two. I am happy to take a stab at this, but I am still very new to robotics controls/estimation and would appreciate any guidance from those of you with years of expertise in this field.

**Only publishing valid data + signal_quality = 1**
https://review.px4.io/plot_app?log=3031d965-adf1-40b5-b9b5-88003973564f
[image]
[image]
**Publishing valid data and invalid data with signal_quality equals 1 or 0 respectively**
https://review.px4.io/plot_app?log=0f0872ec-cf75-4010-913e-cc95fb03c290
[image]
[image]

mhkabir, 2018-08-05:
I believe that the controller height limit kicks in. I'm having issues with FlightPlot, but can you check the `vehicle_local_position.hagl_max` field?

dakejahl, 2018-08-05:
You are correct, looks like `hagl_max` is 2.25. But why does this create an artificial barrier?

mhkabir, 2018-08-06:
Hm yeah it should only be enabled if you're using optical flow and are exclusively relying on it for position estimation. This is in order to stay within operational limits of the PX4flow sensor.

Are you using an optical flow sensor?

dakejahl, 2018-08-06:
Ahh that make sense. We are not using optical flow though :confused:

mhkabir, 2018-08-06:
I can have a look tomorrow.

dakejahl, 2018-08-06:
Thanks, I appreciate that! I have been digging around a bit and can't track it down, the logic is kinda hard to follow in `control.cpp`

dakejahl, 2018-08-07:
@mhkabir I found the culprit. Commented this out and I am no longer seeing the issue.
```
	bool relying_on_rangefinder = _control_status.flags.rng_hgt;

	bool relying_on_optical_flow = _control_status.flags.opt_flow && !(_control_status.flags.gps || _control_status.flags.ev_pos);

	// Do not require limiting by default
	*vxy_max = NAN;
	*vz_max = NAN;
	*hagl_min = NAN;
	*hagl_max = NAN;

	Keep within range sensor limit when using rangefinder as primary height source
	if (relying_on_rangefinder) {
		*vxy_max = NAN;
		*vz_max = NAN;
		*hagl_min = rangefinder_hagl_min;
		*hagl_max = rangefinder_hagl_max;
	}
```
I understand the logic here... If we are using the range finder as our primary height reference we obviously do not want to exceed its limitations. Unfortunately _**range aiding**_ actually switches `_control_status.flags`  from whatever the default is to `rng_hgt` and back to the default again when aiding is no longer determined to be available. I propose we add a new flag (`range_aiding`) to the `filter_control_status_u` struct. This way we are being explicit about our control mode.

mhkabir, 2018-08-07:
Makes sense. However, why is your sensor driver reporting the short max range in the first place? I agree that the switching is not nice, but the controllers are technically correctly helping you stay within reported sensor limits.

dakejahl, 2018-08-07:
I am reporting the max and min ranges because it seemed like the correct thing to do as the `distance_sensor_s` supplies these fields. As someone new (or someone who is casual) it's not practical to have to hunt down the implementation details of the `range_max` and `range_min` values to decide whether or not they should be supplied to the report. It is my opinion that it should be the underlying mechanisms (controllers, estimators, and state machines) that ultimately decide whether those values in those fields actually get used to do something useful (like limit hagl). In the case of **range aiding** the max value is not important (other than throwing away potential junk data, but the driver should be handling that anyways), but nonetheless the field should be populated for code completeness, or in case I decide to turn `EKF2_RNG_AID` off. Let me know your thoughts!

dakejahl, 2018-08-07:
@mhkabir How do you feel about this?

`bool relying_on_rangefinder = _control_status.flags.rng_hgt #&& !_params.range_aid;`

This is all that is necessary to fix this ceiling bug. **Range aiding** implies that the range finder is _aiding_ and not intended to always be relied on. 

See https://github.com/PX4/ecl/pull/491

priseborough, 2018-08-24:
https://github.com/PX4/ecl/pull/491 has been merged

closed 2018-08-24 by priseborough