PX4/PX4-Autopilot #10171: UAVCAN broadcast cost to much cpu.

ToppingXu, 2018-08-06:
**Describe the bug**
When using a uavcan esc, the cpu cost increase a lot, and finally I found it is caused by publisher.broadcast() function. And it is also increased by the mixer output channel count.

**To Reproduce**
Steps to reproduce the behavior:
1.enable uavcan esc.
2. update the mixer output to more than 6 six channel.
3. power on.
4. using top command to see the cpu load.

**Expected behavior**
Can anybody help improve the uavcan driver.

**Log Files and Screenshots**
*Always* provide a link to the flight log file:
https://github.com/PX4/Firmware/blob/master/src/modules/uavcan/uavcan_main.cpp#L971
https://github.com/PX4/Firmware/blob/master/src/modules/uavcan/actuators/esc.cpp#L168

LorenzMeier, 2018-08-06:
Definitely, something we're interested in making more efficient. @davids5 @pavel-kirienko Do you have immediate suggestions?

pavel-kirienko, 2018-08-06:
My immediate suggestion is to profile the firmware using the Poor Man's Sampling Profiler https://github.com/PX4/Firmware/blob/master/platforms/nuttx/Debug/poor-mans-profiler.sh

The sudden CPU load increase at more than 6 ESC is because a large number of channels forces the protocol stack to use multi-frame transfers, which are costlier.

ToppingXu, 2018-08-06:
@pavel-kirienko yes, but we have to use multi-frame transfers, the cpu load increase too much up to 30% and even more, I thought it used a block way to send message on can driver layer.

pavel-kirienko, 2018-08-06:
I'm not saying that multi-frame transfers are to be avoided - they are supported just for the use cases like yours. Let's have a look at the profile and see where to go next. Can you keep the profiler running for an hour or so?

ToppingXu, 2018-08-06:
I am not available here today, and I will do it tomorrow

ToppingXu, 2018-08-07:
@pavel-kirienko So far, I am not able to use flamegraph. 
[image]. Can you give some detail. I do need your help, thanks.

pavel-kirienko, 2018-08-07:
The usage info is provided in the header comment: https://github.com/PX4/Firmware/blob/master/platforms/nuttx/Debug/poor-mans-profiler.sh#L3-L18

The options are not documented but their purpose can be understood just by looking at the code. Sorry, there is no proper documentation for that tool!

ToppingXu, 2018-08-07:
can you show me what command you use. Below is what I used.
[image]

pavel-kirienko, 2018-08-07:
This might help - as it turns out, I did publish some docs but forgot about that: https://dev.px4.io/en/debug/profiling.html

dagar, 2018-08-07:
@ToppingXu @pavel-kirienko I also started adding cmake level helpers. https://github.com/PX4/Firmware/blob/master/platforms/nuttx/CMakeLists.txt#L309-L325

    make px4fmu-v4pro_default profile

ToppingXu, 2018-08-07:
@dagar I use make px4fmu-v4pro_default profile still show "ERROR: No stack counts found
"

dagar, 2018-08-07:
@ToppingXu make sure gdb is setup to connect with the attach in .gdbinit.

ToppingXu, 2018-08-07:
hope this may help you solve the problem

pavel-kirienko, 2018-08-07:
@ToppingXu please attach SVG. Also, how many samples did you record?

hamishwillee, 2018-08-07:
@dagar Re https://github.com/PX4/Firmware/issues/10171#issuecomment-411072443  it is worth updating the docs.

Would I be correct in thinking that
```
make px4fmu-v4pro_default profile
```
1.  gets flamegraph, adds it to the path, then runs the sampler for 1000 points?
2. This replaces all the usage and setup stuff in the docs?
3. This runs on the development computer right, and only on linux/mac (not windows)?

pavel-kirienko, 2018-08-07:
> @dagar <https://github.com/dagar> Re #10171 (comment)

P. 2 not true - this does not replace the usage info. With longer profiling
sessions the script has to be run multiple times with `--append`. Also it
should be possible to change the number of samples.

ToppingXu, 2018-08-08:
@pavel-kirienko I ran 10000 sample point, and there are some error point due to the screen lock process.
the attached is the svg file.

pavel-kirienko, 2018-08-08:
@ToppingXu the image failed to attach properly. Try zipping it first.

ToppingXu, 2018-08-09:
@pavel-kirienko [
[pmpn-flamegraph.svg.zip](https://github.com/PX4/Firmware/files/2272548/pmpn-flamegraph.svg.zip)

](url)

davids5, 2018-08-09:
@pavel-kirienko @ToppingXu use `VERBOSE=1 make <target>` to see what options your build is built with.
 See the MakeFile for [VERBOSE](https://github.com/PX4/Firmware/blob/master/Makefile#L85)

hamishwillee, 2018-08-10:
@pavel-kirienko Thanks for comment on docs above. I have added a new section on basic use, and pushed implementation section (detailed usage) to bottom. Assumption is that it is easier to set up and use the toolchain, and that people can try that quickly then move on to the more detailed usage. 

https://github.com/PX4/Devguide/pull/577

pavel-kirienko, 2018-08-13:
@ToppingXu no, but we could use help with this: https://github.com/UAVCAN/libuavcan_stm32/issues/1

I don't have time to code that right now, but I would happily accept a pull request.

ToppingXu, 2018-08-31:
@pavel-kirienko  You know I am a beginner on uavcan and was not able to change what you mentioned. Can you help when you have time.

closed 2019-02-18 by stale[bot]