PX4/PX4-Autopilot #10172: local position unstable (ekf2 with vision fusion)

virtualclone, 2018-08-06:
Hi, I use ekf2 vision fusion( with motion captures system)
Sometimes, local_position/pose is correct.

But sometimes, it's uncorrect. The pose will be like this:
```
pose: 
  position: 
    x: -1.52471171309e-17
    y: 3.06921519898e-33
    z: 0.124502159655
  orientation: 
    x: 0.00153229214155
    y: 0.0453438805825
    z: 0.0199263977732
    w: -0.998771528692

```
I might reboot the px4 several times to get a correct value of local_position/pose.
What's the problem about this?
My motion capture data streaming rate is at 120Hz.
And I see that the receiving data stream rate of vision_pose/pose is only at 10Hz(in QGroundControl Analyz)
Is that the problem about the large different between the above two data stream rate(120Hz&10Hz)?

EdXian, 2018-08-06:
Hi all,
I have the same problem as the following figuew.
[image]

ChristophTobler, 2018-08-06:
Can you provide a log? A ROSbag with the respective topic would also help

AlexisTM, 2018-08-24:
The "very strange very close to 0 position" is an actual zero that went through multiple conversions and finishes being something.
It happens when the position is lost/EKF Filter fails.

It could be due to: 
* driver: Update frequency
* EKF2: Yaw not correlated and not fused
* WiFi: latency spikes
* ...

priseborough, 2018-08-24:
The estimator status and ekf2 innovation data is not being logged fast enough to determine if data is being lost.

priseborough, 2018-08-28:
@virtualclone Are you able to get full rate logging data? The data rate looks like it is being limited in the setExtVisionData function, but timing jitter and time offsets can still cause problems. Can you drop the rate you are sending data and also get a log of the innovation messages a the full published rate?

kawasakiDaichi, 2018-08-28:
Same with me #10146

kawasakiDaichi, 2018-08-28:
I tried even publish the vision pose at 80 Hz and 10 Hz and didn't make any difference with me,
The EKF keep restarting after a little bit of time

kawasakiDaichi, 2018-08-28:
I think this #10310 is connected to this issue too.

tiralonghipol, 2018-09-03:
Any improvements on this?
following

virtualclone, 2018-09-04:
@priseborough I drop the rate from 50Hz to 10Hz (vision_pose), but the problem is still happened.
Another question, how can I get a log of the innovation messages a the full published rate?

priseborough, 2018-09-11:
@virtualclone Remove the specified time step from this line and it should log every message:

https://github.com/PX4/Firmware/blob/master/src/modules/logger/logger.cpp#L616

ssingh19, 2019-02-05:
Is there any solution to this? I'm getting the same issue.

priseborough, 2019-02-07:
Logs please with any reported problem. Logs need to be from boot and without dropouts that prevent ekf2 replay .

YCH188, 2019-03-03:
@virtualclone @EdXian @ssingh19 Have you solve the problem? I am also using offboard mode to control Pixhawk. When I use rostopic echo /mavros/local_position/pose, the x,y coordinate is close to 0, even I move the drone. But the z coordinate changes as I moved the done. What is the sloution???

mhkabir, 2019-03-03:
Closing this as no one got back with logs. Please open a new issue with logs if you still have issues.

closed 2019-03-03 by mhkabir

ssingh19, 2019-03-03:
@YCH188 I found the issues to be here: https://github.com/PX4/Firmware/blob/82aa24adfca29321cfd1209e287eab6c2b16780e/src/modules/mavlink/mavlink_receiver.cpp#L1239

https://github.com/PX4/Firmware/blob/82aa24adfca29321cfd1209e287eab6c2b16780e/src/modules/mavlink/mavlink_receiver.cpp#L1251

This is where it attempts to sync the vision message timestamp with PX4 clock. As a temporary workaround (which seems to be working reliably - we are able to do pretty aggressive offboard control), [this](https://github.com/StanfordASL/Firmware/commit/b019efd24f8777c2c4d7abda083b61f89191ea23) is our fix. 

You can optionally also incorporate your estimate of the delay between motion capture and the px4 board like [this](https://github.com/StanfordASL/Firmware/commit/1297008abb436b9fdd9f59c0026e57f730bc4b1b#diff-d9c39530a9fff85e8b746162990ca263). 

Note that this last fix is hacky and should be done using the EKF2_EV_DELAY parameter, but you get the idea.

tiralonghipol, 2019-03-04:
@virtualclone @kawasakiDaichi @YCH188 
I solved this problem on the ros side, px4 works perfectly
assuming that your position estimator is ok, try to check the timestamps: I had an almost 500ms difference between my (rovio) pose estimator and the /mavros/local_position/pose
forcing a ros time reference worked for me and the results are amazing!

https://www.youtube.com/watch?v=U-ro2bNXKRk&t=11s

(px4 1.8.2 stable release)
Hope this helps

Seanmatthews, 2019-05-06:
@tiralonghipol Can you expand on how you solved this problem? Did you use the method that @ssingh19 described?