PX4/PX4-Autopilot #10176: MPC_YAW_MODE stuck/not responding to changes in value

santiago3dr, 2018-08-06:
Flew a few flights on a Pixracer (v4) running master d3f5c06b
parameter MPC_YAW_MODE seems to be stuck in mode 3 where the heading points towards the trajectory

First couple flights with the value set to 3 intentionally:
https://logs.px4.io/plot_app?log=32f554d3-c66e-4604-a731-63a4e9e721ff
https://logs.px4.io/plot_app?log=94bc44cd-673c-4c71-bb89-05b50047397d

Changed MPC_YAW_MODE to 0 and observed the same heading:
https://logs.px4.io/plot_app?log=36bb7811-e852-4c0c-9462-fe18aed4e0b1
https://logs.px4.io/plot_app?log=0c8d6dd0-5f19-4c83-b14e-54e57431943b

Changed MPC_YAW_MODE to 1 and observed the same heading:
https://logs.px4.io/plot_app?log=df2c014f-e2c2-4a61-83c3-dcd9c1add6e2
https://logs.px4.io/plot_app?log=f8f03d5c-9c48-49bf-8cef-bc2ac480c478

@Stifael thoughts? Changing the value seemed to work fine in #10003 not sure why it's not working on master after merge

Stifael, 2018-08-08:
The bug occurs because the triplet received from the navigator always contains a valid yaw-setpoint that is set to 0.
[image]

 As I mentioned in https://github.com/PX4/Firmware/pull/10003#issuecomment-409965154, the triplet yaw-sepoint will always take precedence over the `MPC_YAW_MODE`.
 The bug is introduced either by the navigator or the mission actually contains a yaw-setpoint  of 0.

Stifael, 2018-08-08:
@dagar within the navigator, I only find one place where `current.yaw_valid` is set. I assume it is again an issue with bad reset.

santiago3dr, 2018-08-16:
Found the cause
If you import a .waypoint mission into qgroundcontrol, it will be converted into a .plan but not correctly
The issue is does not present itself if running a mission created from scratch in qgroundcontrol as .plan

Made two almost identical missions
First made in .waypoint and converted to .plan where issue is present:
https://logs.px4.io/plot_app?log=940f22c4-6835-46ea-9b44-a3d4aa9e0e6f
https://logs.px4.io/plot_app?log=e49fdf04-e741-423c-a20f-bede459f3592

Second made directly in qgroundcontrol as a .plan where issue is no longer present:
https://logs.px4.io/plot_app?log=607a076c-c0f0-4fa9-a203-98984fab3d73
https://logs.px4.io/plot_app?log=341ec91c-dc51-4db5-bf33-ba15df074e97

closed 2018-08-16 by santiago3dr