PX4/PX4-Autopilot #10177: Set Servo Mission mav command does not seem to work.

tubeme, 2018-08-07:
Hello guys,

We are trying to use Set Servo Mission command without success.

[image]

Here we use channel 3 presuming the channel we use here are AUX channels.

This way on arming the servo goes to 1500 but then on set servo waypoint on ch 3 at 2000us nothing happens. The servo does not release.

We tried all the possibilities for each channel 1-6 once to move to 1000us and once to move 2000us in  mission flights.

Nothing happens the drop servo just does not move at all on set servo mission point.

Are we missing something? I clearly do not see any way I should initialize the payload servo from the mixer.

Any help would be appreciated with this issue. We are in an emergency right now with this drop on target.

We use 1 month old master of QGC and 1.8 PX4 stable.

QGC is Windows version.

Any ideas?

dagar, 2018-08-07:
I wouldn't be surprised if this feature has deteriorated. Do you happen to have a log that can be shared? I want to look for the command followed by any corresponding errors.

minominium, 2018-08-08:
Ummm....dear tubeme, did you solve the problem?

I too expriencing same issue that servo is not functioned in mission mode.

All the settings are just same, except our servo is connected to AUX Pin 1.

And in QGC, we tried with changing AUX Pin, or PWM term.

If you reached the solution, please comment me. Thanks!

kpetrykin, 2018-08-30:
Hello!

Maybe [this pull-request](https://github.com/PX4/Firmware/pull/10320) will help?

GaoGeolone, 2018-12-08:
@tubeme Everything goes well on the drone, but it failed to work on the fixed-wing UAV. Please give me a hand. @kpetrykin

tubeme, 2018-12-10:
@GaoGeolone It does not work in VTOL as well.

GaoGeolone, 2018-12-10:
@tubeme maybe it is the fw_att_control that affect the performance in fixed-wing UAVs when I set servo.
[image]

closed 2019-03-24 by stale[bot]