PX4/PX4-Autopilot #10185: VTOL: Return behaves dangerously in some situations.

tubeme, 2018-08-07:
Hello guys,

I define it as feature request although it has bug connotation to it.

We've had this discussion on the git before about how should the VTOL Return function be implemented.

Four days ago we had a total crash of our plane fully loaded ~10k US.

This was a case where the Return original behavior though consisted somehow crashed the plane.

We had to perform a mission flight at 1000 m height. So we prefer to gain the altitude by hand or in big loiter and when the altitude is gained we activate the mission.
Everything went ok at first. The pilot gained altitude but at that height you barely see the VTOL. He climbed to 1000m and activated the mission. But there was something wrong with the mission. So he decided to abort the mission. During this moments the plane was gone in Position for 30 seconds in one direction. The pilot could not find the vtol in the sky so by telemetry in the QGC he saw where the plane is and decided to activate Return. He told me "I could not see the plane but I heard the motors of the quad ramping and the plane obviously transitioning to MC." 

So imagine in this moment if you want to perform RTL. 1000m descending in MC with 1.5m/s... If it is emergency landing there is no way the battery could hold 1000m.

The pilot then decided to make a transition back to FW flight.  He is not sure what he had done with the controls in this situation, either he removed the Return mode and continued on POS or just flipped the transition switch while in Return mode... The result is that the moment he saw the VTOL it was nosediving in FW flight 80 degrees down to the lake without any control what so ever.

We try to retrieve the plane with divers to recover the SD card with LOGS if possible and SD card with video. No luck so far...

We have the auto-saved .tlog files but the qgroundcontrol is somehow broken and does not want to replay the .tlog files.

The main point is we want to have RTL or Return for the VTOL to behave exactly like an airplane up to the moment it reaches the RTL_DESCEND_ALT altitude. Then it will loiter for the wait period then transition and land. 

If this was the implemented behavior,  in our situation the pilot could activate Return mode the VTOL could have spiraled down to 30 meters probably, make a loiter for the waiting period and then transition to MC and land.

Also in missions if we want to use the Return command at the end of the mission to return the vtol it will perform the return at current altitude and transition, and then start in MC to descend. This makes us to have to program the return with descending points before we land at the end of the mission. If the return is implemented correctly the end of the mission could be just using the RTL point in the route planner.

Because there are people that would like to have the current behavior it might be enabled or disabled with parameter.

At least this is a 10k US lesson learned.

dagar, 2018-08-07:
Sorry to hear about your crash. This feature has been discussed before. One concern is how to do the transition safely after the descend. You don't have any way to specific the direction of the transition, where it starts, or finishes.

The other thing that can be done is to have a safe transition and landing planned at the end of the mission (like a regular plane) and mark the beginning with DO_LAND_START. Then with `RTL_TYPE 1` when you initiate RTL it's actually mission mode starting at DO_LAND_START.

Did you have any of the quadshoot features enabled? These initiate an emergency transition based on altitude loss or exceeding certain extreme angles.

dagar, 2018-08-07:
FYI @RomanBapst @bresch @sanderux

tubeme, 2018-08-07:
I think the fastest and safest method is after the plane holds at loiter alt for 50 seconds... (depends on preset seconds) and it is time to land, to straighten the plane from the loiter and immediately make the transition. This way the plane will move not more than 10-30 meters from the circle. Once in MC the plane could rotate YAW toward the landing point and start moving forward and descending at the the same time until land. When the MC moves forward and descends it does it more stable than just descend and also it spends a lot less energy. So energy wise this will be the most economical way to do it. 

The other way is to make an S maneuver to go out from the loiter then come back and transition at the landing spot and then vertically descend to land.

tubeme, 2018-08-07:
@dagar We do not use the QuadChute yet, because we did not test it. But after this crash we will give more attention to the quadchute.

tubeme, 2018-08-07:
@dagar Also there is another safety matter regarding this behavior.

When I have RTL came back and make the transition then my transition switch stays at airplane because while in Return i do no touch anything. Then If I decide to go out from RTL what is going to happen? Is the vtol momentarily transition in airplane. 

If in Return we have a airplane behavoiur until the moment of landing we have consistency. If I switch off the RTL the plane will straighten its flight the transition switch will be in airplane position so I will continue to fly as usual airplane. Then I could decide to transition back etc.

Antiheavy, 2018-08-12:
> The other thing that can be done is to have a safe transition and landing planned at the end of the mission (like a regular plane) and mark the beginning with DO_LAND_START. Then with RTL_TYPE 1 when you initiate RTL it's actually mission mode starting at DO_LAND_START.

We use RTL_TYPE=1 exclusively with our fixed wing.  Works very well.  I assume it would work just as well for VTOL.

moreba1, 2018-10-10:
I have an idea for RTL. This idea needs to change in autopilot code and the Qgroundcontrol software . we can design a landing pattern such as a fixed wing landing pattern using the software. The UAV must returns to a loiter(altitude) waypoint in FW mode ,Then when it’s reach to direction of home, it must at that altitude start the transition to to MC and reach to the top of the home. Then reduce the altitude to Land .
In this landing pattern we can define this parameters:
1- distance(to Land point) and heading of loiter waypoint
2- loiter altitude
3- landing waypoint

tubeme, 2018-10-27:
Ping ping @RomanBapst @bresch @sanderux @LorenzMeier 

We almost crashed today again.

We hit the geofence, we did not program mission to have a landing pattern and it triggered RTL during flight.

The only thing that saved the plane was that we were flying pretty low at 100 meters.

If we had a 400m with drained battery from the returning flight I'm pretty sure it would have been a crash and a different story.

I this should have a high priority. Much higher than "position controller in transition" i think. 

We are already in a real life situations and applications and this kills machines.

LorenzMeier, 2018-10-27:
@tubeme There are different use cases with different priorities. What you are describing is not really a software fault but something that in your (very particular) operating configuration is understandably an issue. You're more than welcome to contribute the required change or, alternatively, engage with one of the organizations offering commercial support for PX4 to get it added at a faster pace.

There are many other things that need to get done for PX4, many of them affecting a far larger user base. Other companies I know of solve this by having their own developers or by contracting a consultant to contribute it. Just raising it in the issue tracker repeatedly won't get it implemented.

Open source is about collaboratively solving a problem. I'm happy to discuss offline how to best get there. Thanks!

tubeme, 2018-10-28:
@LorenzMeier I understand the whole idea behind the open source because I use Linux since the first Slackware came out in 95 (there was no even RedHat back then), but sometimes some bugs are really critical, and not only to us believe me. All the VTOL projects I've spoken to have this problem and are afraid of engaging RTL... That is the truth, and we lost a $15k craft in a lake because of this. I would rather have given this money to somebody to make it to the whole community...

Thank you for the reply and I will follow your advice to discuss the issues in personal note.

closed 2019-02-11 by stale[bot]