PX4/PX4-Autopilot #10188: REQUEST: Pix4D Mapping improvements needed!

tubeme, 2018-08-07:
Hello guys,

We specialize one of our configurations for mapping so that are the R&D efforts at the moment.

They say a system is as good as its weakest point. 
In other words a system as good as its functional benefits it delivers.

IE if I'm a geodesy pro and want to use drone for mapping I don't care how complex is the drone or how well it flies. I care if the drone can fly only with tablet, it returns home every time and it delivers the geottaging info and images fast and in an integrated way. I put accent to FAST.

Now 1 minute of flight generates about 6mb log data. Our surveys can be as long as 90-120min. It generates about 600-800mb log file.

Ussually the autopilot is berried somewhere in the plane as is the SD card. So the only practical way to download the log file is with USB connection. Now try to download 700mb log with the USB and you will understand my point. 

It is countless and countless hours waiting for log files to download if this is your profession , just to extract from them 7 values at each image trigger point.
Time Lat, Lon, Alt, Pitch, Roll, Yaw Time

I've seen recently couple of chineese systems that do this in a matter of seconds. It can be done during flight even... So despite they are not flying very good, their system is not polished on any side, except the practicality of the purpouse it is meant for.

So here is the request:
Is there a way PX4 can generate on the go a TXT document in each survey and to record each trigger as a text on one line
Trig# Time Lat Lon Alt Pitch Roll Yaw.

This text file will be so small that it could be downloaded in real time through the telemetry, or pushed to the QGC on USB plug or even Push the TXT file through the telemetry when the mission ends.

Then the parser could add the names of the images to the file, or even geotag the images.

I think if the survey generates such a file, there is no need to add the triggers to the log file and make it even more cluttery.
 The TXT will be absolutely enough for survey work.

Of course there should be some development on the QGC side for this to work. @DonLakeFlyer 

This way we could have really practical system for Professional Pix4D work.

Now even the parser in QGC does not provide enough info for Pix4D. https://github.com/mavlink/qgroundcontrol/issues/6772

What do you think?

tubeme, 2018-08-07:
Because Pix4D license is very expensive and also the hardware that is needed for Pix4D to work is very cranked and expensive as well there are many studios that provide Pix4D services.

They need a TXT file with this info and the images. if the TXT is generated automatically from the PX4 in flight, the the parser could use it to add the image names and we are ready to go. Using the TXT will make the workflow much faster and practical.

Antiheavy, 2018-08-07:
I was litterally having this exact same conversation with a customer in the field this afternoon.

tubeme, 2018-08-07:
@Antiheavy We were having lots of this conversations recently. And the worst part is they are opening some local Ground stations on their laptops with local autopilots that only do this but do it brilliantly fast, and decently integrated. So we have to convince them somehow to switch to our system...

LorenzMeier, 2018-08-07:
@Antiheavy @tubeme That was on my list for 1-2 years and we'll get to that shortly. I've jotted down a few thoughts on mission logs - as it is more general I've created a separate feature request:
https://github.com/PX4/Firmware/issues/10189

DonLakeFlyer, 2018-08-07:
Isn't this just more work in QGC to parse the log files to and generate that file? Why would it be firmware side?

tubeme, 2018-08-07:
@DonLakeFlyer Sorry Don I did not understand probably well, but I thought that the Triggers info is generated in the PX4 during flight and logged and is not passed to the QGC... Do we have the triggers from the mavlink in the QGC real time? If so then it is only QGC work...

Antiheavy, 2018-08-07:
Because the existing px4 log file formats are huge and impractical to download.

Antiheavy, 2018-08-07:
If depending on ground side logs you may not have the data from loss of communications events.

DonLakeFlyer, 2018-08-07:
> Because the existing px4 log file formats are huge and impractical to download.

That makes some sense. Some special log file just used for geotagging. Would require new download capability in QGC as well as geotag changes.

DonLakeFlyer, 2018-08-17:
I'm thinking all of this should go here: https://github.com/PX4/Firmware/issues/10189. And the Pix4D should be updated to read whatever this new log format is.

closed 2019-02-18 by stale[bot]