PX4/PX4-Autopilot #10191: Helicopter attitude control law

yuxiqun, 2018-08-08:


yuxiqun, 2018-08-08:
Hello! I want to design an attitude control system of  small traditional helicopter (include  main rotor and tail rotor ). I firstly create a dynamic model of the helicopter, which used to compute total external force and moment and motions states.  Secondly， I use PID to design a MIMO control system, which include attitude control ，altitude control and engine rotation speed. So， I do simulation in MATLAB where I do many PID parameter tunes.But finally I found my helicopter attitude control worse.
     
   I find PX4 use the " mc_attitude_controller" ，that it is multicopter attitude control law, to traditional helicopter.Maybe you can say PX4 use a different mixer between mulicoter and helicoter. My quesition is ,I don't understand why helicopter attitude control can realized in PX4.

dagar, 2018-08-08:
It definitely can be. @bartslinger, anything you'd like to add?

bartslinger, 2018-08-09:
I don't understand the question. Did you try it already on a real helicopter?

yuxiqun, 2018-08-09:
@bartslinger @dagar  I do it in MATLAB Simulation, not on a real helicopter. My question is how the traditional helicopter controlled on the PX4 platform. Thanks for your reply!

bartslinger, 2018-08-10:
It uses the multicopter attitude controller. By selecting the helicopter airframe, it uses a helicopter mixer file to send the right commands to the servos.