PX4/PX4-Autopilot #10195: Is altimeter (LidarLite) compensating the roll/pitch?

tuloski, 2018-08-08:
Is the estimation of the terrain altitude considering the attitude of the UAV?
Any pointer to the code?
What happen if I put the altimeter on a stabilized gimbal such that it always point downward?

CarlOlsson, 2018-08-09:
> Is the estimation of the terrain altitude considering the attitude of the UAV?

Yes, it is [here](https://github.com/PX4/ecl/blob/master/EKF/terrain_estimator.cpp#L122)

> What happen if I put the altimeter on a stabilized gimbal such that it always point downward?

You will need to change the code to not account for the tilt angle

mhkabir, 2018-08-09:
@tuloski please use the forum for further questions.

closed 2018-08-09 by mhkabir

MariliG, 2018-08-09:
Hey @tuloski I found your question very interesting, so I created a post on the forum [here](http://discuss.px4.io/t/is-altimeter-lidarlite-compensating-the-roll-pitch/7450).