PX4/PX4-Autopilot #10197: Possible namespace mismatch between PX4's urtps templates and eProsima FastRTPS headers

L3w1s-L1u, 2018-08-09:
**Describe the bug**
When compiling micrortps_agent code, the compiler will complain missing definition of "MatchingInfo" when generating topic publisher code.

**To Reproduce**
Steps to reproduce the behavior:
Follow the guidance [RTPS/ROS2 Interface: PX4-FastRTPS Bridge](https://dev.px4.io/en/middleware/micrortps.html) to build micrortps_agent
1. using `px4io/px4-dev-nuttx-clang` docker image as build environment
2. clone `PX4/Firmware.git` and `eProsima/FastRTPS.git` repos.
3. [build and install FastRTPS from source](https://dev.px4.io/en/setup/fast-rtps-installation.html)
4. build `px4fmu-v4/5_rtps` target
5. build microRTPS_agent code:
```bash
    # cd build/px4fmu-v4_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/build && cmake  ..
    # make
```
**Expected behavior**
micrortps_agent binary should be built within above build folder

**Log Files and Screenshots**
```bash
Scanning dependencies of target micrortps_agent
[  4%] Building CXX object CMakeFiles/micrortps_agent.dir/optical_flow_PubSubTypes.cpp.o
[  9%] Building CXX object CMakeFiles/micrortps_agent.dir/microRTPS_agent.cpp.o
In file included from /srv/imu.git/build/px4fmu-v4_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/RtpsTopics.h:35:0,
                 from /srv/imu.git/build/px4fmu-v4_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/microRTPS_agent.cpp:47:
/srv/imu.git/build/px4fmu-v4_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/sensor_combined_Publisher.h:68:50: error: 'MatchingInfo' has not been declared
         void onPublicationMatched(Publisher* pub,MatchingInfo& info);
                                                  ^
In file included from /srv/imu.git/build/px4fmu-v4_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/RtpsTopics.h:36:0,
                 from /srv/imu.git/build/px4fmu-v4_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/microRTPS_agent.cpp:47:
/srv/imu.git/build/px4fmu-v4_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/sensor_baro_Subscriber.h:69:52: error: 'MatchingInfo' has not been declared
         void onSubscriptionMatched(Subscriber* sub,MatchingInfo& info);
                                                    ^
In file included from /srv/imu.git/build/px4fmu-v4_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/RtpsTopics.h:37:0,
                 from /srv/imu.git/build/px4fmu-v4_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/microRTPS_agent.cpp:47:
/srv/imu.git/build/px4fmu-v4_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/optical_flow_Subscriber.h:69:52: error: 'MatchingInfo' has not been declared
         void onSubscriptionMatched(Subscriber* sub,MatchingInfo& info);
                                                    ^
CMakeFiles/micrortps_agent.dir/build.make:110: recipe for target 'CMakeFiles/micrortps_agent.dir/microRTPS_agent.cpp.o' failed
make[2]: *** [CMakeFiles/micrortps_agent.dir/microRTPS_agent.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/micrortps_agent.dir/all' failed
make[1]: *** [CMakeFiles/micrortps_agent.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
```
**Additional context**

This error occurred due to "MatchingInfo" been hidden within `rtps` namespace when declaring in headers provided by eProsima thus invisible by PX4's publisher/subscriber template files.

Noticed that PX4 uses templates located in `msg/templates/urtps` instead of provided by upstream `eProsima/FastRTPS.git` so am I missing any update to keep urtps templates fresh?

Also noticed that urtps templates' license info are newer than eProsima source( I'm using HEAD of master). If urtps' templates are correct then the instructions in [dev.px4.io](https://dev.px4.io/en/setup/fast-rtps-installation.html) should be updated, otherwise urtps' template should be patched.

TSC21, 2018-08-09:
@L3w1s-L1u this is an ongoing work. I will update you as soon as possible.

TSC21, 2018-08-09:
Oh now that I see: You don't seem to be using the latest changes done to the urtps templates. What's your Fimrware git commit tag? I am almost sure about this because you have `void onSubscriptionMatched(Subscriber* sub,MatchingInfo& info);` (no whitespace before `MatchingInfo`), while in upstream: https://github.com/PX4/Firmware/blob/master/msg/templates/urtps/Subscriber.cpp.template#L100.

What you are reporting above was already addressed here: https://github.com/PX4/Firmware/pull/9643. About the documentation, as said, it's being updated, though this issue is not related to outdated docs (they lack some clarifications but they still apply to this case).

closed 2018-08-09 by TSC21

L3w1s-L1u, 2018-08-12:
@TSC21 Thanks for clarifying this and sorry for misreported this as an issue. Should have done some thorough search about 'urtps' before opening an issue.

Yes I'm using a local commit branched from Firmware V3.3 stable release, and maybe that's a lit bit stale with regard to the urtps part.  I'll switch my work onto a branch which already contains that namespace fix.