PX4/PX4-Autopilot #10198: How to call aes_decrypt() function in commander.cpp?

xdwgood, 2018-08-09:
@bkueng  
I added `#include <nuttx/crypto/aes.h>` in commander.cpp.
And use this `aes_decrypt()` function in commander.cpp

But I encountered this error at the end of make.
**xdw@xdw:~/src/Firmware$ make tap-v3_default** 
```
[1/3] Linking CXX executable nuttx_tap-v3_default.elf
FAILED: : && /home/xdw/gcc-arm-none-eabi-5_4-2016q2/bin/arm-none-eabi-g++   -fno-common -ffunction-sections -fdata-sections -mcpu=cortex-m7 -mthumb -mfpu=fpv5-sp-d16 -mfloat-abi=hard -fno-common -ffunction-sections -fdata-sections -g -fno-exceptions -fno-rtti -std=gnu++11 -fno-threadsafe-statics -DCONFIG_WCHAR_BUILTIN -D__CUSTOM_FILE_IO__ -fcheck-new -Wall -Warray-bounds -Wdisabled-optimization -Werror -Wextra -Wfatal-errors -Wfloat-equal -Wformat-security -Winit-self -Wlogical-op -Wmissing-declarations -Wmissing-field-initializers -Wpointer-arith -Wshadow -Wuninitialized -Wunknown-pragmas -Wunused-variable -Wno-unused-parameter -Wunused-but-set-variable -Wformat=1 -Wdouble-promotion -Wno-missing-field-initializers -Wreorder -fvisibility=hidden -include visibility.h -fno-strict-aliasing -fomit-frame-pointer -funsafe-math-optimizations -ffunction-sections -fdata-sections -fno-strength-reduce -fno-builtin-printf -Os -DNDEBUG  -mcpu=cortex-m7 -mthumb -mfpu=fpv5-sp-d16 -mfloat-abi=hard -nodefaultlibs -nostdlib -Wl,--warn-common,--gc-sections src/firmware/nuttx/CMakeFiles/nuttx_tap-v3_default.elf.dir/__/__/platforms/empty.c.obj  -o nuttx_tap-v3_default.elf -L/home/xdw/src/Firmware/build/tap-v3_default/external/Install/lib -T/home/xdw/src/Firmware/build/tap-v3_default/NuttX/nuttx/configs/tap-v3/scripts/ld.script -Wl,-Map=/home/xdw/src/Firmware/build/tap-v3_default/nuttx_tap-v3_default.map -Wl,--warn-common -Wl,--gc-sections -Wl,--start-group msg/libuorb_msgs.a src/drivers/device/libdrivers__device.a src/drivers/stm32/libdrivers__stm32.a src/drivers/stm32/adc/libdrivers__stm32__adc.a src/drivers/stm32/tone_alarm/libdrivers__stm32__tone_alarm.a src/drivers/led/libdrivers__led.a src/drivers/px4fmu/libdrivers__px4fmu.a src/drivers/boards/common/libboards_common.a src/drivers/boards/common/stm32/libboard_stm32.a src/drivers/boards/tap-v3/libdrivers__boards__tap-v3.a src/drivers/rgbled_pwm/libdrivers__rgbled_pwm.a src/drivers/tap_esc/libdrivers__tap_esc.a src/drivers/mpu6000/libdrivers__mpu6000.a src/drivers/ms5611/libdrivers__ms5611.a src/drivers/mpc2520/libdrivers__mpc2520.a src/drivers/hmc5883/libdrivers__hmc5883.a src/drivers/ist8310/libdrivers__ist8310.a src/drivers/hc_sr04/libdrivers__hc_sr04.a src/drivers/gps/libdrivers__gps.a src/drivers/airspeed/libdrivers__airspeed.a src/drivers/realsense/libdrivers__realsense.a src/modules/sensors/libmodules__sensors.a src/drivers/vmount/libdrivers__vmount.a src/drivers/camera_trigger/libdrivers__camera_trigger.a src/drivers/gimbal_protocol_splitter/libdrivers__gimbal_protocol_splitter.a src/drivers/mavlink_dup/libdrivers__mavlink_dup.a src/systemcmds/bl_update/libsystemcmds__bl_update.a src/systemcmds/led_control/libsystemcmds__led_control.a src/systemcmds/mixer/libsystemcmds__mixer.a src/systemcmds/param/libsystemcmds__param.a src/systemcmds/perf/libsystemcmds__perf.a src/systemcmds/pwm/libsystemcmds__pwm.a src/systemcmds/hardfault_log/libsystemcmds__hardfault_log.a src/systemcmds/motor_test/libsystemcmds__motor_test.a src/systemcmds/reboot/libsystemcmds__reboot.a src/systemcmds/top/libsystemcmds__top.a src/systemcmds/config/libsystemcmds__config.a src/systemcmds/nshterm/libsystemcmds__nshterm.a src/systemcmds/mtd/libsystemcmds__mtd.a src/systemcmds/dumpfile/libsystemcmds__dumpfile.a src/systemcmds/ver/libsystemcmds__ver.a src/systemcmds/sd_bench/libsystemcmds__sd_bench.a src/systemcmds/tap_esc_config/libsystemcmds__tap_esc_config.a src/systemcmds/topic_listener/libsystemcmds__topic_listener.a src/systemcmds/tune_control/libsystemcmds__tune_control.a src/modules/commander/libmodules__commander.a src/modules/events/libmodules__events.a src/modules/load_mon/libmodules__load_mon.a src/modules/navigator/libmodules__navigator.a src/modules/mavlink/libmodules__mavlink.a src/modules/land_detector/libmodules__land_detector.a src/modules/ekf2/libmodules__ekf2.a src/modules/mc_att_control/libmodules__mc_att_control.a src/modules/mc_pos_control/libmodules__mc_pos_control.a src/modules/vtol_att_control/libmodules__vtol_att_control.a src/modules/logger/libmodules__logger.a src/modules/sdlog2/libmodules__sdlog2.a src/modules/systemlib/param/libmodules__systemlib__param.a src/modules/systemlib/libmodules__systemlib.a src/modules/systemlib/mixer/libmodules__systemlib__mixer.a src/modules/uORB/libmodules__uORB.a src/modules/dataman/libmodules__dataman.a src/lib/controllib/liblib__controllib.a src/lib/mathlib/liblib__mathlib.a src/lib/mathlib/math/filter/liblib__mathlib__math__filter.a src/lib/ecl/libecl.a src/lib/geo/liblib__geo.a src/lib/geo_lookup/liblib__geo_lookup.a src/lib/conversion/liblib__conversion.a src/lib/launchdetection/liblib__launchdetection.a src/lib/led/liblib__led.a src/lib/terrain_estimation/liblib__terrain_estimation.a src/lib/runway_takeoff/liblib__runway_takeoff.a src/lib/tailsitter_recovery/liblib__tailsitter_recovery.a src/lib/version/liblib__version.a src/lib/DriverFramework/framework/liblib__DriverFramework__framework.a src/lib/rc/liblib__rc.a src/lib/tunes/liblib__tunes.a src/lib/FlightTasks/liblib__flight_tasks.a src/platforms/nuttx/libplatforms__nuttx.a src/platforms/common/libplatforms__common.a src/platforms/nuttx/px4_layer/libplatforms__nuttx__px4_layer.a NuttX/apps/libapps.a NuttX/nuttx/arch/arm/src/libarch.a NuttX/nuttx/binfmt/libbinfmt.a NuttX/nuttx/binfmt/libbinfmt.a NuttX/nuttx/libc/libc.a NuttX/nuttx/configs/libconfigs.a NuttX/nuttx/libxx/libcxx.a NuttX/nuttx/drivers/libdrivers.a NuttX/nuttx/fs/libfs.a NuttX/nuttx/mm/libmm.a NuttX/nuttx/sched/libsched.a -Wl,--end-group -lm -lgcc ROMFS/libromfs.a src/lib/DriverFramework/framework/src/libdf_driver_framework.a && :
src/modules/commander/libmodules__commander.a(commander.cpp.obj): In function `commander_thread_main(int, char**)':
/home/xdw/src/Firmware/build/tap-v3_default/../../src/modules/commander/commander.cpp:2165: undefined reference to `aes_decrypt'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
Makefile:155: recipe for target 'tap-v3_default' failed
make: *** [tap-v3_default] Error 1
```

Is this because the function under nuttx cannot be accessed directly? what should I do? thank you very much :heart:

xdwgood, 2018-08-09:
I added all the functions to commander.cpp and solved the problem temporarily. Passed the compilation.
But repeating the copy code, this is not good :cry:

dagar, 2018-08-09:
@xdwgood the nuttx library and linking is probably missing. 

Try adding nuttx crypto here (like the networking example) - https://github.com/PX4/Firmware/blob/master/platforms/nuttx/NuttX/CMakeLists.txt#L196-L209

Then here for linking 
 - https://github.com/PX4/Firmware/blob/master/platforms/nuttx/CMakeLists.txt#L51-L66

dagar, 2018-08-15:
Is this resolved?

xdwgood, 2018-08-16:
@dagar   I don't know if I understand what you mean.
I added https://github.com/PX4/Firmware/blob/master/platforms/nuttx/NuttX/CMakeLists.txt#L196-L209 to https://github.com/PX4/Firmware/blob/master/platforms/nuttx/CMakeLists.txt#L51-L66, and the final result is not compiled.

> Is this resolved?
No.

xdwgood, 2018-08-21:
@dagar 
I know the way to solve the problem through the internal staff and your help. Thanks, close it. :smiley:

closed 2018-08-21 by xdwgood