PX4/PX4-Autopilot #10203: CI Failure on Jenkins Firmware Compile: No space left on device

AlexisTM, 2018-08-09:
**Describe the bug**

The Jenkins Firmware Compile CI system fails on the PRs, for example: #7477 

http://ci.px4.io:8080/blue/organizations/jenkins/PX4_misc%2FFirmware-compile/detail/PR-7477/2/pipeline#step-375-log-32

```bash
Submodule 'src/modules/uavcan/libuavcan' (https://github.com/UAVCAN/libuavcan.git) registered for path 'libuavcan'
Cloning into 'src/modules/uavcan/libuavcan'...
fatal: write error: No space left on device
fatal: index-pack failed
fatal: clone of 'https://github.com/UAVCAN/libuavcan.git' into submodule path 'src/modules/uavcan/libuavcan' failed
Cloning into 'src/modules/uavcan/libuavcan'...
fatal: write error: No space left on device
fatal: index-pack failed
fatal: clone of 'https://github.com/UAVCAN/libuavcan.git' into submodule path 'src/modules/uavcan/libuavcan' failed
CMake Error at src/modules/uavcan/CMakeLists.txt:54 (add_subdirectory):
  add_subdirectory given source "libuavcan" which is not an existing
  directory.
```

dagar, 2018-08-09:
Thanks for reporting this as a bug. We're a little sloppy with temp scratch space because these machines are dynamically provisioned on ec2 as need and then automatically recycle on idle.

dagar, 2018-08-09:
I've increased the threshold for required free disk space before a build slave is taken offline (and then terminated).

closed 2018-08-09 by dagar