PX4/PX4-Autopilot #10218: Does Pixhawk support this 20 motor mixing?

karl900, 2018-08-12:
Hi all!
I am new here and I am very sorry if I am in the wrong section or asking the wrong question.
I am looking for a suitable FC and software to support a custom mixing of this following shape : 20  rotors as seen in the imaged linked to below: 

https://discuss.ardupilot.org/uploads/default/original/3X/1/4/14ad874d4c0e434571f614067dc5b6505efefa29.jpg

Can the Pixhawk do really custom motor mixing in the first place ? The answer is not clear when I search the webs. Again I apologize for any inconveniences. If anyone would guide  me I would be truly thankful! 

Thanks
Regards
/Karl

dagar, 2018-08-12:
Hi Karl,

Yes that's doable if you can get all the outputs attached to the same flight controller. For example take a look at the generic dodecarotor coaxial airframe configuration. https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d/24001_dodeca_cox

In this case it's split into 2 mixers  (top & bottom) to distribute across a pixhawk's MAIN and AUX outputs.

karl900, 2018-08-14:
We connect 10 motors to the 5 OUTPUTS and the 10 motors to the other 5 AUX?  like this I assume?:
[https://ibb.co/kvKb89](url)

# @output MAIN1 motor 1 + 2                        
# @output MAIN2 motor 3 + 4
# @output MAIN3 motor 5 +6                       UPPER MOTORS
# @output MAIN4 motor 7 + 8
# @output MAIN5 motor 9 + 10
#
# @output AUX1 motor 11+12
# @output AUX2 motor 13 +14
# @output AUX3 motor 15 + 16                     LOWER MOTORS
# @output AUX4 motor 17 +18
# @output AUX5 motor 19 +20

This way we have 5 upper pairs, where each upper pair are connected to same  OUTPUT channel). and 5 lower pairs where each pair are  connected to the same AUX channel.
Yaw however could be the only problem? what are other problems here?:)
Am I understanding this  almost correctly? 

Thanks
Regards
/Karl

closed 2019-02-18 by stale[bot]