PX4/PX4-Autopilot #10220: SITL orb poll failing intermittently

dagar, 2018-08-12:
http://ci.px4.io:8080/blue/organizations/jenkins/PX4_misc%2FFirmware-SITL_tests/detail/pr-nuttx_adc_dma/2/pipeline/55/
[image]

http://ci.px4.io:8080/blue/organizations/jenkins/PX4_misc%2FFirmware-SITL_tests/detail/pr-nuttx_adc_dma/2/pipeline/57
[image]

dagar, 2018-08-12:
Possibly related https://github.com/PX4/Firmware/pull/10030

LorenzMeier, 2018-08-13:
Can we please do all temp files if possible in working folder.

closed 2018-08-13 by bkueng

dagar, 2018-08-13:
> @dagar do jenkins build run in parallel and is the /tmp directory shared between the instances?

No, to keep it simple it's always 1 job per machine.

PR #10224 (2018-08-13): Fix posix shell startup race
The FIFO was created in the server thread, and the PX4 main thread could already have continued and started to execute the bash script. In that case the client tried to open the FIFO but it did not exist yet.

Client error as seen in CI (I did not reproduce it):
```
ERROR [px4_daemon] pipe open fail
ERROR [px4_daemon] Could not send commands
```

Fixes #10220