PX4/PX4-Autopilot #10223: Flight Tasks Issues

RomanBapst, 2018-08-13:
**Description**
1) When releasing the x/y stick in position control mode the drone decelerates as it should but there is always a twitch in attitude. You can see in the log that for some reason the velocity setpoint pulls towards the current velocity.
[image]

2) For some reason it seemed like our parameter changes did not take effect. We e.g. changed the parameter MPC_XY_VEL_MAX from 2 to 8 but the drone would not move faster than 2 m/s.
The parameter changes were logged so it was not a communication issue. 

We were on this commit from master:
30fb82d9b4157b74c741ec7e1e16d7408e5dfa3f

Log file:
https://logs.px4.io/plot_app?log=798b9ed5-d68c-4c22-99a8-3b3540386f50

@Stifael @MaEtUgR Do you have an idea what went wrong? I tested flight tasks a week before on a different vehicle and I had no such issues.

Stifael, 2018-08-15:
@bresch in principle the attitude setpoint can even be discontinuous in normal position control mode without smoothing (MPC_POS_MODE = 0). There is nothing that prevents the attitude setpoint from having step inputs except the slew-rate stuff. However, even the slew-rate stuff acts on the velocity level and not on the thrust vector level with which the attitude is computed from. 

@RomanBapst, the `MPC_XY_VEL_MAX` is basically the maximum allowed speed for the entire system. If you want to decrease the speed for manual, then you should set `MPC_VEL_MANUAL` to 2. 

(as a side note: the comment for `MPC_XY_VEL_MAX` is wrong: `MPC_XY_VEL_MAX` does not only apply to auto, but rather for the entire system when position control is enabled).

That said, the itch is somewhat weird, but I suspect it is related to the smoothing stuff. 
@RomanBapst can you set `MPC_POS_MODE` to 0? This will turn off the smoothing. If that itch is still present, then it is the slew-rate logic.

RomanBapst, 2018-08-15:
@Stifael Thanks, I'll give that a try and will get back to you.

RomanBapst, 2018-11-06:
Fixed in #10745

closed 2018-11-06 by RomanBapst