PX4/PX4-Autopilot #10228: SITL default does not start with latest master

ecmnet, 2018-08-14:
Starting SITL with 'make posix_sitl_ekf2 jmavsim' does not start SITL correctly.
Refer to the shell output below:

sitl_bin: /Users/ecmnet/PixHawk/src/Firmware/build/posix_sitl_ekf2/bin/px4
rcS_path: posix-configs/SITL/init/ekf2
debugger: none
program: jmavsim
model: none
src_path: /Users/ecmnet/PixHawk/src/Firmware
build_path: /Users/ecmnet/PixHawk/src/Firmware/build/posix_sitl_ekf2
**empty model, setting iris as default**
SITL COMMAND: /Users/ecmnet/PixHawk/src/Firmware/build/posix_sitl_ekf2/bin/px4 /Users/ecmnet/PixHawk/src/Firmware/ROMFS/px4fmu_common -s etc/init.d-posix/rcS -t /Users/ecmnet/PixHawk/src/Firmware/test_data
usage: grep [-abcDEFGHhIiJLlmnOoqRSsUVvwxZ] [-A num] [-B num] [-C[num]]
**Error: Unknown model 'iris'**
	[-e pattern] [-f file] [--binary-files=value] [--color=when]
	[--context[=num]] [--directories=action] [--label] [--line-buffered]
	[--null] [pattern] [file ...]
INFO  [Unknown] Creating symlink /Users/ecmnet/PixHawk/src/Firmware/ROMFS/px4fmu_common -> /Users/ecmnet/PixHawk/src/Firmware/build/posix_sitl_ekf2/tmp/rootfs/etc

______  __   __    ___

dagar, 2018-08-14:
Thanks for reporting, things are going to be a little messy while the startup is unified across platforms.

Try `make posix_sitl_default jmavsim`. The per estimator targets will be gone soon. The posix_sitl_config config already contains all of them. https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d-posix/rcS#L65-L72

ecmnet, 2018-08-14:
`make posix_sitl_default jmavsim` returns the same error, however.

dagar, 2018-08-14:
Are you on a Mac? I tried it on Ubuntu. It looks like a problem with grep usage here https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d-posix/rcS#L44.

[image]

ecmnet, 2018-08-14:
Yes, on a Mac...

PR #10236 (2018-08-15): posix rcS: use sed instead of grep -P
The -P argument seems not to be supported on OSX

@ecmnet can you test this?

Fixes #10228
closed 2018-08-15 by dagar