PX4/PX4-Autopilot #10230: Yaw setpoint jumps sharply

mwingerson, 2018-08-14:
I am having an issue with a very heavy prototype VTOL that I'm building and I am seeing an issue that I am not sure where this behavior could be coming from. As I am hovering, sometimes the bird yaws hard and loses altitude. I think the fact I lose altitude indicates to me that we don't have enough thrust to push through the decent caused by spooling down half of the motors in the yaw.

While I have been digging through the logs, I noticed a point where the yaw setpoint jump sharply at 7:36. I also noticed a baro EKF error and would that case an EKF reset on the mags which could cause a sharp heading change?

I feel like this might just be a poor calibration and tune but I thought I would ping you to see if you have any great ideas.

https://review.px4.io/plot_app?log=ca81fd27-ac49-4d8f-aac2-da9e1532396d

dagar, 2018-08-15:
What's the closest upstream PX4 commit?

mwingerson, 2018-08-22:
I'm working off of commit: 92ff391c9765a394fdf3ef81d0e9bc6985ef8f07

The only changes that I have made are to create a custom mixer so I shouldn't have affected the core functionality. After some more digging, it looks like we are running into issues trying to use the Here+ magnetometer as an external sensor instead of the internal mags that are buried in the bird. 

I found a couple other threads on getting a working driver for the Here+ GPS but I couldn't find whether there is a definitive fix or not.

Thanks!

closed 2019-02-18 by stale[bot]