PX4/PX4-Autopilot #10233: problem with dual gps, or GPS2 port

KeyneZzz, 2018-08-15:
my hardware is pixhawk2 cube, and i'm trying the newly merged dual gps  function #9765 
when i connected 2 gps on both gps ports, it seems not working properly
i logged it and can see there's no vehicle_gps_position_1 or 2 in it, only 0
so i doubted the one on GPS2 port not working
when i'm only connecting ONE GPS on GPS2 port, QGC report no gps
so i think maybe i miss something when configuring the GPS2 port, can anyone help? 

thanks

KeyneZzz, 2018-08-15:
https://logs.px4.io/plot_app?log=d777c91c-4fdd-4da2-8a08-26aceaad6c44

here's my log

KeyneZzz, 2018-08-15:
i reviewed the changed file and find the ekf_gps_position message and the selected variety. it is 0 in my log. is it the switcher key? and how to modify it?

priseborough, 2018-08-15:
By default PX4 does not look for a second GPS. 

You will need to modify the ROMFS/px4fmu_common/init.d/rcS file so that the following start command is used to start the GPS:

gps start -d /dev/ttyS3 -e /dev/ttyS <GPS2 serial port number>

I don't know what serial port GPS2 connects to with your hardware.

KeyneZzz, 2018-08-16:
@priseborough thanks a lot. i've tried successfully.

priseborough, 2018-08-16:
Can you let us know what serial device numbers you ended up using? TXS

LorenzMeier, 2018-08-16:
@bkueng I've assigned you to this - can we please add an enum in QGC to configure? Thanks!

bkueng, 2018-08-16:
Requires https://github.com/PX4/Firmware/pull/9615

KeyneZzz, 2018-08-17:
@priseborough i'm referrring to this page http://ardupilot.org/copter/docs/common-pixhawk-serial-names.html

closed 2019-02-18 by stale[bot]