PX4/PX4-Autopilot #10235: How to handle custom flight tasks?

ChristophTobler, 2018-08-15:
There should be a possibility to add a custom flight task (and launch it) without much effort or creating a large diff and conflicts.

Our long term goal is to move the logic if a task can run/be activated to the actual flight task because we think the current situation with commander knowing everything doesn't scale.

*Approaches*

- adding a custom state (https://github.com/PX4/Firmware/pull/10199)

dagar, 2018-08-15:
One idea was to leave room in the state machine for custom classes of nav state.
 - custom fully autonomous
 - custom manual modes with full position control 
 - etc

These buckets would allow you to do anything you want within with the same control mode flags, but ensure consistent failsafe handling.

dagar, 2018-08-15:
Conveying the mode to the ground is another problem.

ChristophTobler, 2018-08-16:
>One idea was to leave room in the state machine for custom classes of nav state.

Could be a short term option

>#10237 is a good example of why this decentralized handling is potentially a bad idea. We need to have this logic handled by well defined state machines, but that doesn't mean all within commander.

I think the current situation is a good example of why the commander should not have everything predefined.
Yes, there needs to be a centralized handling of tasks and fail-safes but with feedback from the respective tasks and not hard-coded into a state machine.

closed 2019-02-18 by stale[bot]