PX4/PX4-Autopilot #10239: Can we check GCC/compiler version in builds?

hamishwillee, 2018-08-15:
AFAIK we only "guarantee" against a particular version of GCC. It would be good if the toolchain could warn against incorrect GCC, and possible consequences.

In particular, I'm hoping that a warning might reduce the number of people who complain about FMUv2 out of flash issues because they aren't using the right version.

dagar, 2018-08-15:
We could do this afternoon a toolchain update so that the actual GCC version will be known (major, minor, patch). 

That being said, I suspect people are going to miss/ignore the warning, and I think an actual failure would be too strict.

hamishwillee, 2018-08-15:
Yes, a failure would be too strict. 

You're right that most people will miss the warning. Still, I tend to read through those things when things go wrong.

Up to you though.

dagar, 2018-08-15:
Let's add the warning and see if anyone notices. The toolchain update here provides access to the compiler version. https://github.com/PX4/Firmware/pull/9808

closed 2019-02-18 by stale[bot]