PX4/PX4-Autopilot #10243: Gazebo on Ubuntu 16.04 fails

DonLakeFlyer, 2018-08-15:
Been like this for me of over a week or so. Running from latest master:
```
make distclean
make posix gazebo_typhoon_h480

```
```
[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 10.0.0.35
[Wrn] [msgs.cc:1852] Conversion of sensor type[sonar] not supported.
[Err] [Plugin.hh:180] Failed to load plugin libgazebo_gst_camera_plugin.so: libgazebo_gst_camera_plugin.so: cannot open shared object file: No such file or directory
[Msg] Camera on udp port 14530
[Wrn] [gazebo_sonar_plugin.cpp:84] [gazebo_sonar_plugin] Please specify a robotNamespace.
[Dbg] [gazebo_mavlink_interface.cpp:132] joint [rotor_4_joint] found for channel[0] joint control active for this channel.
[Dbg] [gazebo_mavlink_interface.cpp:132] joint [rotor_5_joint] found for channel[1] joint control active for this channel.
[Dbg] [gazebo_mavlink_interface.cpp:132] joint [rotor_2_joint] found for channel[2] joint control active for this channel.
[Dbg] [gazebo_mavlink_interface.cpp:132] joint [rotor_3_joint] found for channel[3] joint control active for this channel.
[Dbg] [gazebo_mavlink_interface.cpp:132] joint [rotor_0_joint] found for channel[4] joint control active for this channel.
[Dbg] [gazebo_mavlink_interface.cpp:132] joint [rotor_1_joint] found for channel[5] joint control active for this channel.
[Dbg] [gazebo_mavlink_interface.cpp:132] joint [typhoon_h480::cgo3_camera_joint] found for channel[6] joint control active for this channel.
[Dbg] [gazebo_mavlink_interface.cpp:132] joint [typhoon_h480::cgo3_camera_joint] found for channel[7] joint control active for this channel.
[Dbg] [gazebo_mavlink_interface.cpp:132] joint [typhoon_h480::cgo3_vertical_arm_joint] found for channel[8] joint control active for this channel.
[Dbg] [gazebo_mavlink_interface.cpp:132] joint [left_leg_joint] found for channel[9] joint control active for this channel.
[Dbg] [gazebo_mavlink_interface.cpp:132] joint [right_leg_joint] found for channel[10] joint control active for this channel.
10.10
[Msg] Using MAVLink protocol v2.0
[Msg] GimbalControllerPlugin::Init
[Err] [Publisher.cc:96] EXCEPTION: Invalid message type

terminate called after throwing an instance of 'gazebo::common::Exception'
SITL COMMAND: /home/parallels/repos/Firmware/build/posix_sitl_default/bin/px4 /home/parallels/repos/Firmware/ROMFS/px4fmu_common -s etc/init.d-posix/rcS -t /home/parallels/repos/Firmware/test_data
INFO  [Unknown] Creating symlink /home/parallels/repos/Firmware/ROMFS/px4fmu_common -> /home/parallels/repos/Firmware/build/posix_sitl_default/tmp/rootfs/etc
67 WARNING: setRealtimeSched failed (not run as root?)
```

yl5006, 2018-08-16:
I   have   the same problem ，may be gazebo version difference，When I disable the Gimbal plugin in typhoon_h480.sdf .  then  compile well

DonLakeFlyer, 2018-08-18:
> When I disable the Gimbal plugin in typhoon_h480.sdf . then compile well

@yl5006 Thanks for the workaround. This gets me going again.

hamishwillee, 2018-11-12:
FYI I updated the title. This still fails on latest Firmware, but it only applies to Typhoon/Camera plugin, not to "all of gazebo". 

@dogmaphobic Who is right person to look at this one?

Haya-Alsharif, 2018-11-14:
The solution for this error message `(not run as root?)` is to simply add `sudo` before running the command.

hamishwillee, 2018-11-14:
@Haya-Alsharif Thanks - I don't see forcing people to run make in sudo as a solution :-) @bkueng Who is best person to talk to about the gazebo gimbal plugin?

hamishwillee, 2018-11-15:
@yl5006 That's great! If you're fairly sure can you please create a PR - that is easier for people to test and review.

yl5006, 2018-11-15:
I have aready pull a request in sitl_gazebo  module ，can some one review？

hamishwillee, 2018-11-15:
Perhaps add cross link to here for that request?

hamishwillee, 2018-11-15:
Thanks. No , the submodule id gets updated periodically.

hamishwillee, 2018-11-19:
> Hello @hamishwillee , I think the problem is permission.
> @Haya-Alsharif Thanks - I don't see forcing people to run make in sudo as a solution :-)
> But actually sudo helps to create .so files some OS.

There is nothing else on ubuntu that requires sudo to build the PX4 toolchain so there is no point in me trying it - because even if it works, it is not IMO a good solution.

But I'm not a developer. @bkueng is there someone who should "own this"?

bkueng, 2018-11-19:
> But I'm not a developer. @bkueng is there someone who should "own this"?

Maybe @TSC21 can help?
And yes, sudo is not the way to go.

Haya-Alsharif, 2018-11-19:
@bozkurthan exactly, this is how I got it to work.
@hamishwillee Here is what I meant by `sudo`: The error in the code above is caused by the fact that `libgazebo_gst_camera_plugin.so` gazebo model or its px4 mixers are missing. This can be caused by one of the following reasons:
1) The entire model located at `Firmware/tools/sitl_gazebo/models/typhoon_h480/` is missing or is not added to `$GAZEBO_MODELS_PATH` and `$GAZEBO_PLUGIN_PATH`. The **solution** is to make sure you have them under the specified directory and add `export GAZEBO_...` commands to your `~/.bashrc` to make sure nothing override them.

2) A symlink between your built folder directory `/Firmware/build/posix_sitl_default/tmp/rootfs/etc` and the directory containing the model's mixers located at `/Firmware/ROMFS/px4fmu_common/init.d-posix/6011_typhoon_h480` was not created due to the lack of `sudo` privileges, which was indicated by the shared error message below. A **simple fix** is to create those links only _once_ using `sudo` and then the warning message can be neglected and won't result in your code termination. **Another solution** is obviously to create that link manually. 
```
SITL COMMAND: /home/parallels/repos/Firmware/build/posix_sitl_default/bin/px4 /home/parallels/repos/Firmware/ROMFS/px4fmu_common -s etc/init.d-posix/rcS -t /home/parallels/repos/Firmware/test_data
INFO  [Unknown] Creating symlink /home/parallels/repos/Firmware/ROMFS/px4fmu_common -> /home/parallels/repos/Firmware/build/posix_sitl_default/tmp/rootfs/etc
67 WARNING: setRealtimeSched failed (not run as root?)
``` 
Side note: most of the installation instruction for px4 assumes that you are the `sudo` user, for example, you would see something like "install the pacakge: `apt-get install blablabla`", with no `sudo`. 
-- Haya

hamishwillee, 2018-11-21:
There is a huge difference between setting up *installation packages* that require `sudo` and making our build process depend on it. But as I said, I'm not a developer and I won't be attempting to fix this.  @TSC21 Assigning to you to find best person to own this :-)

Jaeyoung-Lim, 2019-09-26:
I am seeing this problem again on master with the following command.
` make px4_sitl_default gazebo_typhoon_h480`
Out put is as the following.
```
[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.34.54
[Wrn] [msgs.cc:1852] Conversion of sensor type[sonar] not supported.
[Wrn] [Event.cc:61] Warning: Deleting a connection right after creation. Make sure to save the ConnectionPtr from a Connect call
[Err] [Plugin.hh:180] Failed to load plugin libgazebo_gst_camera_plugin.so: libgazebo_gst_camera_plugin.so: cannot open shared object file: No such file or directory
[Msg] Camera on udp port 14530
[Wrn] [gazebo_sonar_plugin.cpp:75] [gazebo_sonar_plugin] Please specify a robotNamespace.
[Wrn] [gazebo_barometer_plugin.cpp:66] [gazebo_barometer_plugin] Using default home altitude of 488 m
[Dbg] [gazebo_mavlink_interface.cpp:129] joint [rotor_4_joint] found for channel[0] joint control active for this channel.
[Dbg] [gazebo_mavlink_interface.cpp:129] joint [rotor_5_joint] found for channel[1] joint control active for this channel.
[Dbg] [gazebo_mavlink_interface.cpp:129] joint [rotor_2_joint] found for channel[2] joint control active for this channel.
[Dbg] [gazebo_mavlink_interface.cpp:129] joint [rotor_3_joint] found for channel[3] joint control active for this channel.
[Dbg] [gazebo_mavlink_interface.cpp:129] joint [rotor_0_joint] found for channel[4] joint control active for this channel.
[Dbg] [gazebo_mavlink_interface.cpp:129] joint [rotor_1_joint] found for channel[5] joint control active for this channel.
[Dbg] [gazebo_mavlink_interface.cpp:129] joint [typhoon_h480::cgo3_camera_joint] found for channel[6] joint control active for this channel.
[Dbg] [gazebo_mavlink_interface.cpp:129] joint [typhoon_h480::cgo3_camera_joint] found for channel[7] joint control active for this channel.
[Dbg] [gazebo_mavlink_interface.cpp:129] joint [typhoon_h480::cgo3_vertical_arm_joint] found for channel[8] joint control active for this channel.
[Dbg] [gazebo_mavlink_interface.cpp:129] joint [left_leg_joint] found for channel[9] joint control active for this channel.
[Dbg] [gazebo_mavlink_interface.cpp:129] joint [right_leg_joint] found for channel[10] joint control active for this channel.
[Msg] Connecting to PX4 SITL using TCP
[Msg] Lockstep is enabled
[Msg] Speed factor set to: 1

```

TSC21, 2019-09-26:
@Jaeyoung-Lim `gazebo_gst_camera_plugin` will only be compiled if gstreamer is found on your system: https://github.com/PX4/sitl_gazebo/blob/master/CMakeLists.txt#L358. That's why it is not finding the the shared library file.

Jaeyoung-Lim, 2019-09-26:
@TSC21 It turned out to be a problem with the `GAZEBO_PLUGIN_PATH` not being properly set

closed 2019-09-26 by Jaeyoung-Lim