PX4/PX4-Autopilot #10244: snapdragon801  ERROR [Unknown] Error opening startup file, does not exist: etc/init.d/rcS

zhaoxiaowei1013, 2018-08-16:
when  i run 
```
./px4 mainapp.config
```
```
INFO  [Unknown] Creating symlink mainapp.config -> /home/linaro/etc
ERROR [Unknown] Error opening startup file, does not exist: etc/init.d/rcS
```
```
-- PX4 VERSION: v1.8.0-879-g01744a9ef
-- CONFIG: posix_eagle_default
```

yl5006, 2018-08-16:
this is PATH problem, relative path

RomanBapst, 2018-08-16:
Maybe related to posix shell work of @bkueng

RomanBapst, 2018-08-16:
https://github.com/PX4/Firmware/pull/10173

bkueng, 2018-08-16:
> Maybe related to posix shell work of @bkueng

Correct. You need to make sure `${PX4_SOURCE_DIR}/ROMFS` is uploaded as well (here: https://github.com/PX4/Firmware/blob/master/platforms/posix/CMakeLists.txt#L49) and change the start command to something like `./px4 -s mainapp.config`
@zhaoxiaowei1013 @yl5006 can you try that?
@julianoes fyi

zhaoxiaowei1013, 2018-08-16:
@bkueng  when i use ```./px4 -s mainapp.config```
it works ok, thanks.

closed 2018-08-16 by zhaoxiaowei1013