PX4/PX4-Autopilot #10245: How exactly use MAV_CMD_DO_SET_SERVO command in QGC

minominium, 2018-08-16:
Hi all. I hope you had a nice summer.

I open this issue because I need to use servo motor in mission mode, but it doesn't work..

[image]

this is what I want to do, especially moving servo motor in waypoint mission.

I use SG90 servo motor with additional voltage regulator, and set signal pin to AUX OUT 1.

and in mission command, I set param1 : 1, param2 : 1300

The servo motor is working fine with controller in manual(or stabilized) mode, but I need mission mode.

So what I need to change or set? Any reply will help me :)

QGC : 3.3.2 in windows
PX4 : Latest stable
Frame : 450 sized custom quad

kpetrykin, 2018-08-30:
Hello!

Maybe [this pull-request](https://github.com/PX4/Firmware/pull/10320) will help you?

minominium, 2018-08-30:
@kpetrykin 
Thank you for your response. I'll give it a shot!

closed 2019-02-18 by stale[bot]