PX4/PX4-Autopilot #10251: Add Autonomous Soaring Feature

IeiuniumLux, 2018-08-16:
Add the Autonomous Soaring feature to allow flying wings to respond to rising air current (thermals) in order to extend endurance and gain altitude with minimal use of the motor.

This feature will be mainly for glider type aircraft with a good lift to drag ratio, and be equipped with an airspeed sensor.  Folding props will be a plus too.

### Scenario:
When the Soaring functionality is enabled for a mission, then the plane should begin gliding once has reached a waypoint altitude and it's moving toward the next waypoint. If it reaches the minimum SOAR altitude before arriving to the next waypoint, then throttle is automatically provided and the plane should begin to climb to the altitude of the next waypoint.  Repeat all over again through the mission.

kaklik, 2020-04-30:
Hello, I noticed this issue, which seems to be unresolved. Although it is pointing to Atlantiksolar project.  This project is also mentioned in [user guide documentation](https://docs.px4.io/master/en/frames_plane/) without any description. Does it Atlantiksolar is somehow related to PX4 or it does ever contributed to PX4 code?

junwoo091400, 2023-03-04:
I'm quite interested in this topic, and for now was roaming around these resources:

https://github.com/samuelctabor/Soaring_simulation

https://github.com/Jaeyoung-Lim/mav-soaring

https://github.com/akolobov/ArduSoar-3.8

junwoo091400, 2023-03-04:
Also this research is quite interesting: https://onlinelibrary.wiley.com/doi/abs/10.1002/rob.21765

kaklik, 2023-03-05:
Hi Junwoo @junwoo091400, 

I did some experiments on how to resolve that in the last year.  Our notes are in this in [this issues](https://github.com/ThunderFly-aerospace/PX4Firmware/issues/35). Sorry for the Czech version of the issue, but essentially it is all about how to get the vertical wind velocity estimate. 

I think estimating vertical wind velocity (e.g. thermal) is an essential feature of every "soaring algorithm" applicable to UAVs. The reason is that UAV has a mixed glider/motor mode, where we need to know the "strength" of the thermal even in case of motor running. The links you posted have minimal applicability in reality because they expects pure glider airframe or an airframe that is a capable to transform to engine-free flight.

junwoo091400, 2023-03-05:
> essentially it is all about how to get the vertical wind velocity estimate.

Very interesting read! It does break down to polar curve estimation & wind estimation.

I would assume that for now we would already benefit from a simple soaring controller with motors off, that can simplify the problem for now. Perhaps @tstastny or @Jaeyoung-Lim could have a working version from the past projects?

Because predicting the vertical wind reliably is quite difficult right? @bresch @priseborough. But, I would be interested to know what it would take to accomplish it as well.

kaklik, 2023-03-12:
> But, I would be interested to know what it would take to accomplish it as well.

Ok, I do try to write a summary of my knowledge of this theme:

The issue could be relatively easily solved in case if there could be information about the dissipation of energy (total energy rate) in real-time during the flight versus the energy rate, supplied by the engine. The energy rate supplied by the engine (similar to engine power) should be probably a linear function of the throttle. But we need to know the correct scale (e.g. Throttle_to_STE_rate) to calculate that. It should be something like the multiplicative inverse of [STE_rate_to_throttle](https://github.com/PX4/PX4-Autopilot/blob/3fe4c6e3f56d68417918454188a3f968c1e5ccd6/src/lib/tecs/TECS.cpp#L289) used to control of the engine by the throttle. 

Then in cases where total energy changes during the flight without the influence of throttle, it should be only an external effect that could not be other than vertical airmass velocity.  
There exists a slight complication because the conversion between kinetic and potential energy has losses, which is not in the time of my investigation calculated in TECS. But these losses should be within the practical flight limits and should be a quadratic function of the airspeed. 

Therefore to solve that there is a need to constantly fit the few TECS parameters (rate of energy loss by steady flight at specific airspeed and power contributed by the engine at specific throttle - e.g. throttle to energy rate curve) during flight to actual airframe performance. These parameters should be initialized before the flight by a constant similar to the expected flight performance.  

I think these steps are mandatory because I do not see much simplification by attempt to test the soaring controller at an airframe, with the motor off. It is because the thermals need to be explored at significant heights and also in a significant area, where is a problem to determine a point where we should switch off the motor and start the exploration of the thermals. (especially in a situation with an attempt to tune the soaring algorithm there will be "too many moving parts" to tune)

Additionally the soaring algorithms I expect to be computationally very intensive (based on my current knowledge of some search algorithms from TF-ATMON system). Therefore there is not much sense to try to debug those algorithms on resources-limited flight controllers or onboard companion computers. To avoid that situation telemetry data which could be used on the ground by soaring algorithms is needed. The vertical or Z wind velocity seems to be the obvious choice for me.  The vertical airmass speed should be transmitted to the control station by the usual telemetry link and processed to explore the thermals. Then this is only a step from a very useful robotics paraglider flight assistant. :)