PX4/PX4-Autopilot #10254: PX4 SITL Firmware Start Error - No Such File or Directory

ganscool86, 2018-08-17:
Hi,
I am trying to setup PX4 SITL setup for Airsim. I followed the procedure stated here. First I installed the prerequisites (Bash on Windows and Linux Toolchains) and then pulled the PX4 firmware code and build it using the following commands:
git clone https://github.com/PX4/Firmware.git
cd Firmware
git submodule update --init --recursive
make posix_sitl_default

The build went well without any error message. But when I try to start the PX4 firmware using the following command:
./build/posix_sitl_default/px4 ./posix-configs/SITL/init/ekf2/iris
 or
 ./build_posix_sitl_default/src/firmware/posix/px4 ./posix-configs/SITL/init/lpe/iris

It just shows No such file or directory. Can someone tell me what i am missing and how to get it to work. Thanks.

lamping7, 2018-08-17:
Do those files exist? I thought they were generated with `make posix_sitl_default sitl_gazebo`

hamishwillee, 2018-08-20:
Also you might try building and launching using the full make command for your target (for information on options see http://dev.px4.io/en/simulation/#startingbuilding-sitl-simulation ). That "should" ensure that everything needed is built (and you'll be able to work out where)

hamishwillee, 2018-08-21:
Did this solve the problem?

mikeisted, 2018-10-22:
I have the same issue following instructions at https://github.com/Microsoft/AirSim/blob/master/docs/px4_build.md.
The folder firmware does not contain the folder posix.
I'm working with Ubuntu 16.04.

Appreciate the advice from hamishwillee, but it's not clear what target is required (tried none_iris), but the posix folder is still not created.

hamishwillee, 2018-10-23:
@mikeisted @ganscool86 Sorry, forgot about this until saw @mikeisted post.

I don't know the answer for sure, but I am fairly sure you can _probably_ make and start this using the make command below:
```bash
make posix_sitl_default none_iris  
# Replaces: ./build/posix_sitl_default/px4 ./posix-configs/SITL/init/ekf2/iris
```

Try that please.

hamishwillee, 2018-10-23:
@bkueng Can I get some advice here please. 
1. The airsim instructions previously stated that SITL ekf2/iris should be started as
   ```
   ./build/posix_sitl_default/px4 ./posix-configs/SITL/init/ekf2/iris
   ```
   I have suggested that the equivalent is:
   ```
   make posix_sitl_default none_iris 
   ```
   - Can you confirm (this looks about right according to [Simulation](https://dev.px4.io/en/simulation/) and [System Startup](https://dev.px4.io/en/concept/system_startup.html))?
   - There do not appear to be any ekf2 targets. 
     - Does that mean that all targets are ekf2?
     - Does that mean there is no way to set lpe in SITL via `make`?
1. Now that I've tried this, would I be correct in assuming that:
   - the valid posix targets are the files listed in https://github.com/PX4/Firmware/tree/master/ROMFS/px4fmu_common/init.d-posix - and that the list of targets from `make posix list_vmd_make_targets` is generated from the names in that folder?
   - that the nuttx only targets from the [init.d](https://github.com/PX4/Firmware/tree/master/ROMFS/px4fmu_common/init.d) folder can't be used. What would you do if you did want to support these? Is it more than just copy and modify an init file from init.d to init.d-posix?
1. Is there any reason to manually start a file using `./build/posix_sitl_default/bin/px4` directly and feeding in the init file? Or is the plan to always start using `make` now?

hamishwillee, 2018-11-12:
@bkueng Ping

bkueng, 2018-11-12:
> Can you confirm (this looks about right according to Simulation and System Startup)?

Looks about right to me as well.

> Does that mean there is no way to set lpe in SITL via make?

Not via `make`, no.

> the valid posix targets are the files listed in https://github.com/PX4/Firmware/tree/master/ROMFS/px4fmu_common/init.d-posix - and that the list of targets from make posix list_vmd_make_targets is generated from the names in that folder?

They are defined here: https://github.com/PX4/Firmware/blob/master/platforms/posix/cmake/sitl_target.cmake#L51

> is all that is required for a simulator target a file in init.d-posix? ... or is there more required (I assume so - ie that this is just part and you also need stuff in Jmavsim and in gazebo, airsim, whatever.

The model from the cmake above will be passed to the simulator (here: https://github.com/PX4/Firmware/blob/master/Tools/sitl_run.sh).

> Is there any reason to manually start a file using ./build/posix_sitl_default/bin/px4 directly and feeding in the init file? Or is the plan to always start using make now?

`make` is the official way to go for now, because you need to make sure that some other things are correct as well, such as the working directory. Having a separate script with a simple interface would be better though.

jasonhbartlett, 2018-11-20:
Novice beginner for all this but the documentation needs to be corrected since I also had this issue and had to find the solution by search of the error and coming to this page.  Recommend at least a mention of this issue at https://github.com/Microsoft/AirSim/blob/master/docs/px4_build.md doc.

bkueng, 2018-11-21:
@jasonhbartlett can you confirm that `make posix_sitl_default none_iris` works?

closed 2019-03-05 by stale[bot]

owner01, 2019-10-02:
This issue still seems to be here when trying to do the gazebo simulation

lamping7, 2019-10-02:
As stated above the build structure is very different now than in the original issue description. You'll need to provide more information of the problem.

DavidCarreonJr, 2019-10-15:
> @jasonhbartlett can you confirm that `make posix_sitl_default none_iris` works?

I have ran into this problem while building it out myself. This indeed worked for me. Thank you

sam-rke, 2020-11-03:
i am facing such issues
i am trying to include "esc.hpp" in one of example file if i am giving full path its working fine but i have to give full path in all further included files for example in file esc.hpp we have an include "uavcan/uavcan.h" if i am building code without including "esc.hpp" in example file it's building without any problem.
But if i am trying to include "esc.hpp" in example file first i have to five full path of esc.hpp secondly i have to modify all included files from esc.hpp. like from "uavcan/uavcan.h" to "full_location/uavcan/uavcan.h" and so on in all files....do someone have any idea why this thing is happening??