PX4/PX4-Autopilot #10258: [Global position] unable to obtain lock despite good sat count

santiago3dr, 2018-08-17:
Running current master 57052ed and it took several attempts and almost 3 minutes to obtain global position, successful flights:
https://logs.px4.io/plot_app?log=641838bc-50ff-4654-997f-24cba2347631
https://logs.px4.io/plot_app?log=a3837f58-7045-4539-9270-2f9b1d7a525c

Turned on logging from boot and after 3+ minutes, was unable to obtain global position:
https://logs.px4.io/plot_app?log=ca259319-8446-4dbf-abc0-d55f073387bf

Reverted to yesterday's master 36403e9 and could obtain lock under 30 seconds as normal:
https://logs.px4.io/plot_app?log=79330666-0f92-4cbb-917d-4dc13f66a20f
https://logs.px4.io/plot_app?log=872eb963-c9ef-40a6-8d44-dc53d8941704

@dagar @priseborough thoughts?

mhkabir, 2018-08-17:
Suspect : https://github.com/PX4/Firmware/pull/9999

priseborough, 2018-08-19:
GPS reported speed accuracy indicates poor S/N noise ratio and is failing the revised speed accuracy threshold:

[image]

[image]

For future reference, check the  value of estimator_status_0.gps_check_fail_flags and compare agains the message definition in the estimator_status.msg file to see which check is failing. Then adjust the corresponding threshold threshold. in this instant the value of EKF2_REQ_SACC will need to be increased to 0.7 for this receiver combination. If this was an outdoor test with open sky, then the GPS installation is marginal. We can increase the threshold as indicated at the expense of allowing some setups to start using GPW when indoors which was the problem raised by 
https://github.com/PX4/Firmware/issues/9549 that 
https://github.com/PX4/Firmware/pull/9999 was attempting to fix.

priseborough, 2018-08-19:
Is this older Ublox 7 or 8 hardware? and what else is there in the vehicle that could be causing GPS noise?

dagar, 2018-08-19:
Sidenote - this is another good example of how the interaction with the GCS needs to be improved to reflect what's actually going on in the vehicle. 

Even starting with some kind of large overall "ready to fly" indicator based would be a big improvement.
https://github.com/PX4/Firmware/issues/8697

FYI @DonLakeFlyer @dogmaphobic @antiheavy

DonLakeFlyer, 2018-08-19:
Maybe this is what I was hitting as well when my GPS positioning was bad with my cube.

Antiheavy, 2018-08-19:
@dagar I'm a big supporter of https://github.com/PX4/Firmware/issues/8697

I worry about over-conservative sensor checks though.  https://github.com/PX4/Firmware/pull/9999  We seem to constantly be playing whack-a-mole on sensor and estimator thresholds that are set tight with good intentions for multicopters, but are totally inappropriate for hand launched fixed wings.

dagar, 2018-08-20:
I agree it can be a huge annoyance, and to be honest it's likely going to get worse before it gets better.

I might be wrong, but in my mind a huge portion of the pain is the disconnect between vehicle and ground. If these things were clearly (and calmly/passively) communicated with actionable suggestions then I don't think it would be nearly as bad. Trying to arm shouldn't feel like rolling the dice to see what horrible terse error message you're going to get thrown back next.

Secondly I think we need to consistently make a deliberate effort to reconsider defaults for all vehicle types as these things evolve.

priseborough, 2018-08-20:
@dagar Is there a way we can set up a fixed wing vs rotary wing set of default parameters that doesn't require modifying very platform specific file in the ROMFS/init.d directory?

What constitutes an acceptable GPS quality is application specific and I agree with @Antiheavy 's remarks wrt to  'whack-a-mole'

dagar, 2018-08-20:
At the moment only the vehicle type init scripts. https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d/rc.fw_defaults#L22-L25

It sounds like we'll need to come up with something better. Perhaps we should start with the hybrid VTOL case and work back.

santiago3dr, 2018-08-20:
@priseborough seems it was a hardware issue, i replaced the gps module and ran a few flights on master without issues
https://review.px4.io/plot_app?log=6c1875b0-d206-4b58-b835-00057136e0ff
https://review.px4.io/plot_app?log=bf30e00f-20d4-4f38-a817-84d0715b3de3

closed 2019-02-18 by stale[bot]