PX4/PX4-Autopilot #10262: offboard failsafe to loiter, but the yaw is not stable

KeyneZzz, 2018-08-18:
i set my failsafe mode to hold, and with no RC, it did turn to hold, but i.m guessing the yaw setpoint is unstable so the drone can't hold its heading, just turning at random. 

i searched the problem and it's said that bad gps can cause the problem, but when i tried HITL simulation with accurate GPS information it still appeared, so is there any better way to hold the heading in loiter mode?

dagar, 2018-08-19:
@KeyneZzz which version of PX4 exactly?

@Stifael @MaEtUgR

KeyneZzz, 2018-08-20:
https://logs.px4.io/plot_app?log=fcc6faa6-eea3-48c2-afad-b8b958824258
here's the log

KeyneZzz, 2018-08-20:
https://logs.px4.io/plot_app?log=0e13fde7-88d6-4395-ba0d-649b474abb5a

this is another trial with more recent version on master, and it is acting like it used to be: when turn to falisafe hold, yaw turn to 0. but this is still not what i want. i want it to keep the heading, not turning north.

i have succeeded to modify it in v1.6.x  but now since the pos controller is refactored, i can't find the exact lines to modify it.

KeyneZzz, 2018-08-22:
https://logs.px4.io/plot_app?log=58f3d121-cc66-4b11-9436-3ec3136b23c3

tried the change in the PR above(manually changed), but the problem remains. 

it seems like when turn to offboard-loss failsafe, the navigator should turn to loiter rather than to rcloss.

KeyneZzz, 2018-08-22:
well i find the problem, in mission_block.cpp MissionBlock::set_loiter_item, when initialize the loiter item, the yaw is set to NAN

MaEtUgR, 2018-10-30:
If this NAN yaw setpoint gets through the flight task and to the position control it just means you don't care about the yaw and it will stay uncontrolled e.g. drifting to wherever the drone wants. Does this only happen in failsafe or also when you manually switch to loiter in flight?

Do you want to create a pr for your fix if that solves the upstream bug?

MaEtUgR, 2018-10-30:
I see it's here: https://github.com/PX4/Firmware/pull/10291

MaEtUgR, 2018-10-30:
Still not solved, ongoing here: https://github.com/PX4/Firmware/pull/10297

MaEtUgR, 2019-01-29:
@KeyneZzz Can you please chack and update if this issue still exists?

I just tested switching from Manual Position mode with full yaw rotation into Hold and it locks as expected and switching from arbitrary states in Acro mode into Hold also works fine. There should not be anything different when coming from offboard unless Hold is commanded in a different way. I switched to Hold by pressing the software mode switch in QGC.
@KeyneZzz How did you switch to hold? Could it be that it was via a MAVLink message and there was a certain parameter set differently or did you set the parameter `MPC_YAW_MODE` different from the default?

KeyneZzz, 2019-01-30:
@MaEtUgR In my occasion, the HOLD mode is triggered when it was in OFFBOARD mode but  the offboard control msgs were cut off. Now it's working perfectly.

closed 2019-01-30 by KeyneZzz

MaEtUgR, 2019-01-30:
@KeyneZzz Thanks for checking again 👍