PX4/PX4-Autopilot #10264: Tfmini rangefinder detection have huge delay when use  the telem1 is enable as a radio link

eleboss, 2018-08-19:
**Describe the bug**
**Tfmini rangefinder detection have huge delay when use  the telem1 is enable as a radio link .**

When using the Tfmini rangefinder at telem2 with default setting and using radio link in telem1 at the same time, the rangefinder will not work well, and the range detection will drop to from a very high rate to about 2s per detection. This can be directly view from QGC Analyze.

**To Reproduce**
Steps to reproduce the behavior:
1. Start the pixhawk4
2.plug the radio link in telem1 and tfmini in telem2
3. enable tfmini in param setting
4. use the radio link and QGC analyze can see a huge delay of range detection.
5.unplug the radio link the detection rate will back to normal.

**Expected behavior**
The tfmini should work in about 100hz rate.

**Log Files and Screenshots**

**Drone (please complete the following information):**
- Pixhawk4 with 1.8.0 PX4 FIRMWARE.

dagar, 2018-08-19:
Where are you checking the rate? It sounds like what you're seeing is the distance sensor rate over mavlink. To see the actual driver publication rate try `listener distance_sensor` or `uorb top` on the console (qgc mavlink shell or direct serial console).

eleboss, 2018-08-19:
Sorry about the missing description, I also checked the range data by rostopic echo mavrros/distance_sensor*  in the onboard tx2. and can see the delay when radiolink is working.

eleboss, 2018-08-19:
Also the update of data(unplug the radio link), does not look like100hz(many frame have same range when move the drone up and down very quick), compared with the data visualize from directly connect the tfmini to my windows computer and read data from the official software provide by benewake.

I can post picture is needed.

Antiheavy, 2018-08-20:
I have also noticed TFmini has huge latency , 1-2 seconds, when reporting in QGC analyze widget. Haven’t figured out why.

Antiheavy, 2018-08-21:
> It seems this is why the EKF has been rejecting my range data.

ahhh... no wonder we weren't getting good height results in our test flights:
https://review.px4.io/plot_app?log=68b730e9-c95c-4f0d-afdb-99191e66e35c

fyi @kjkinney

mhkabir, 2018-08-21:
I guess something is wrong with the driver.

mhkabir, 2018-08-23:
Comparing with the baro estimate, my data seems to be around 600ms delayed : 

[image]

With the 600ms delay manually compensated in the logs, the baro and distance sensor estimates align : 
[image]

mhkabir, 2018-08-23:
I see the same 600ms delay in @Antiheavy's logs as well, though it's less obvious.

@bkueng Any idea what could be causing this in the driver? https://github.com/PX4/Firmware/blob/master/src/drivers/distance_sensor/tfmini/tfmini.cpp

bkueng, 2018-08-24:
The driver reads at most 10 bytes per cycle (https://github.com/PX4/Firmware/blob/master/src/drivers/distance_sensor/tfmini/tfmini.cpp#L538). If the device is sending faster, data will start to buffer.
The driver should read all available data.

eleboss, 2018-08-24:
Is there any way to make it just send the latest data？

mhkabir, 2018-08-24:
@bkueng Looks like sensor should send 9 bytes for each measurement, at 100Hz. We should probably run the driver slightly faster than the 100Hz to avoid buffering I believe.
[image]

For the correct fix, is there a design pattern somewhere in Firmware which I can look at? Currently all the UART sensor drivers are implemented like this.

mhkabir, 2018-08-24:
Fix is here : https://github.com/PX4/Firmware/pull/10319

eleboss, 2018-08-25:
Issues closed.

PR #10319 (2018-08-26): tfmini : fix scheduling, parsing and modernize output
This PR :
1. Reduces the driver scheduling interval to 9ms, such that the driver is always ready to read new data. Running it at exactly 100Hz is not correct since the driver and sensor measurement intervals are not "in sync", causing the driver to miss data. This caused a fill-up of the UART buffer.
2. Rejigs parsing logic to always publish the latest measurement. This gets rid of the 600ms delay we had been seeing, since the driver was always parsing old data from the UART buffer, fixed at 10 bytes per every collect call.
3. Removes usele [... cut]
closed 2018-08-25 by eleboss