PX4/PX4-Autopilot #10268: Support for WS2812 LEDs

mzahana, 2018-08-19:
For general questions please use [PX4 Discuss](http://discuss.px4.io/) or [Slack](http://slack.px4.io/).

**Description**
It would help to have clear visual indication/feedback from the drone using clear LEDs, e.g. using a WS2812 LED strip.

**Required**
An implementation of a driver for WS2812  to support the following options would be great.

Possible user-specified options:
* Predefined modes: 
    - For example, to show the same status as the main LED.
    - Other predefined modes, e.g. showing GPS state, battery level, others?
* MAVLink controlled: For example, offboard commands for specific LED settings

The user would do the following,
* Set a parameter to activate the driver
* Set a parameter to choose between a predefined mode, or MAVLink controlled mode
* Set a parameter to choose which AUX pins to use (Will need to check against conflicts, e.g. camera trigger interface)
* Set the number of LEDs to be used

I can help writing the initial version of the driver, but will need some guidance on where to start.

Would it be a **separate** module in [src/drivers](https://github.com/PX4/Firmware/tree/master/src/drivers), or included inside [src/drivers/px4fmu/](https://github.com/PX4/Firmware/tree/master/src/drivers/px4fmu)?

Thanks.

iZhangHui, 2018-08-20:
Great idea! A separate module would be better for my personal view.

mzahana, 2018-08-20:
@iZhangHui Could you clarify the pros/cons of implementing it as separate module or inside px4fmu?
Thanks,

mzahana, 2018-08-23:
@LorenzMeier Any hints?
Also what are the function that can be used to control the pulses sent to the WS2812 LEDs via AUX OUT pins?

mzahana, 2018-08-23:
@proficnc Thanks for the note!

I still think it would not harm to have support for WS2812 type of LEDs as they are already popular and cheap, and is being supported by racing controllers. I believe the recent Pixhawk controllers have even better CPU clocks.

iZhangHui, 2018-08-29:
Driver codes should be put into src/drivers. High level codes can send control messages to WS2812 through uorb. @mzahana

mzahana, 2018-08-29:
I might be wrong, but after looking into the WS2812 protocol, its tight timing requirements may make the scheduler busy which affects the more important flight controls. That is not desirable I guess. Let's see what alternative @proficnc can offer.

A work around for the WS2812 is to have it controlled by an Arduino and the send higher level command from PX4 with the DO_SERVO_MOUNT

closed 2019-02-18 by stale[bot]