PX4/PX4-Autopilot #10270: How to use the mavlink_receiver and mavlink_messages for UAVs communication?

MatteoTadiello, 2018-08-20:
I need to broadcast and receive the gps mavlink message to implement a collision avoidance in a multiple drones application. How can I use this informations in a module?

hamishwillee, 2018-08-20:
Thanks for closing.

That said, I'll provide a starting point, because there are no good documents on this yet.
1. There is discussion of some obstacle avoidance stuff in this (and linked) PRs: https://github.com/PX4/Devguide/issues/530
2. The MAVLink messages for obstacle avoidance are named with `TRAJECTORY_*` - e.g. [TRAJECTORY_REPRESENTATION_WAYPOINTS](https://mavlink.io/en/messages/common.html#TRAJECTORY_REPRESENTATION_WAYPOINTS). PX4 is already doing "stuff" with these - basically telling missions that they need to take a different course, and what that course is.
3. More generally, the [MAVLink messaging](http://dev.px4.io/en/middleware/mavlink.html) doc explains how you can add custom messages (essentially how things work). This even uses a fictional trajectory message for the example.

MatteoTadiello, 2018-08-20:
Ok thank you very much!