PX4/PX4-Autopilot #10271: Cygwin-based Windows Simulation and pkill

hamishwillee, 2018-08-20:
There are various scripts we have that shut down the simulator, and which assume pkill is supported:
```
Tools/sitl_multiple_run.sh: line 18: pkill: command not found
```
Is there any way to either support pkill, or fake this?

hamishwillee, 2018-08-20:
Specifically, this did not work in my first test: https://github.com/Dronecode/DronecodeSDK/issues/494#issuecomment-413602890 POSSIBLY because I had started the simulator twice, and I suspect the old version may not have been cleaned up properly.

MaEtUgR, 2018-08-22:
pkill and the shutdown command works with when installing the correct cygwin packages. Here's the change for the script installation:
https://github.com/PX4/windows-toolchain/commit/0719053e6d057e3c9c3cbf63fcea4802c884a670

It will be in the next cygwin toolchain msi (0.4) which I'll generate when the jmavsim on master is fixed again #10098 .

MaEtUgR, 2018-09-23:
Closing because it's now in version 0.4: https://github.com/PX4/Devguide/pull/614

closed 2018-09-23 by MaEtUgR