PX4/PX4-Autopilot #10272: Parameter descriptions "repetition" necessary?

hamishwillee, 2018-08-20:
Consider https://docs.px4.io/en/advanced_config/parameter_reference.html#SDLOG_PROFILE

The comment includes detailed information about each bit: .e.g.
```
...
0 : Set to true to use the default set (used for general log analysis) 1 : Set to true to enable full rate estimator (EKF2) replay topics 2 : Set to true to enable topics for thermal calibration (high rate raw IMU and Baro sensor data) 3 : Set to true to enable topics for system identification (high rate actuator control and IMU data) ...
```
Then a short form:
```
Bitmask:
0: default set (log analysis)
1: estimator replay (EKF2)
2: thermal calibration
3: system identification
4: high rate
5: debug
6: sensor comparison
```
How is the information used? Ie can we make the short description longer such that we don't need the repetition in the comment? 

This is not a big deal, but I find the information in the comment hard to parse.

dagar, 2018-08-20:
I'd say less is more in this case.

[image]

[image]

hamishwillee, 2018-08-27:
@dagar Looking at this, we have at most 45 chars for the bit descriptions. So we can cut the comment back if we are happy to lose the information that in the brackets.

The minimum we can do losing no information is: #10334. With minimal data loss, we might drop the descriptions to the below - but perhaps this is a step too far.
```
 * @bit 0 Default set (general log analysis)
 * @bit 1 Estimator replay (EKF2)
 * @bit 2 Thermal calibration (high rate raw IMU/Baro)
 * @bit 3 System identification (high rate actuator/IMU)
 * @bit 4 High rate (fast maneuver analysis)
 * @bit 5 Debug (debug_*.msg topics or custom code)
 * @bit 6 Sensor comparison (low rate raw IMU, Baro, Magnetomer)
```

More generally though this is no longer as easy to read.

hamishwillee, 2018-08-27:
@DonLakeFlyer Does the comment field of a parameter respect any formatting?

Specifically, I'd like to be able to mark up text as bullets in our source and have it render as such in QGC parameter editor. Is that possible? The goal is to make some of the comments easier to read.

DonLakeFlyer, 2018-08-27:
> Does the comment field of a parameter respect any formatting?

Nope. But theorettically could be added. But I'd be worried about how any existing text might end up strange if it happened to have formatting in it by accident.

> Alternatively, perhaps enable wrapping of checkbox text?

This is doable. But not currently supported.

hamishwillee, 2018-08-27:
Thanks @DonLakeFlyer . I've raised as a request in https://github.com/mavlink/qgroundcontrol/issues/6833

hamishwillee, 2019-02-04:
Closing. I think we'll just have to consider tidying text on a case-by-case basis

hamishwillee, 2019-02-04:
I'm closing. I'd like to solve this in a systemic way, but I think we'll just have to look at each case separately.

closed 2019-02-04 by hamishwillee