PX4/PX4-Autopilot #10278: DO_REPOSITION command not working

DonLakeFlyer, 2018-08-20:
I'm controlling a drone from QGC first telling it to takeoff and then telling it to rotate to a heading of 0. The takeoff works fine, but the rotate using DO_REPOSITION does not rotate the vehicle. Flight logs are here: https://review.px4.io/plot_app?log=0704e9d3-04ac-49fb-82d1-ee6e46bed798. I can tell from QGC and replaying the flights logs that the DO_REPOSITION command does change the flight mode to HOLD. But it never rotates the vehicle.

All of the above works flawlessly in Gazebo SITL testing.

LorenzMeier, 2018-08-20:
@bkueng @RomanBapst Could you please help to unblock Don? Thanks!

DonLakeFlyer, 2018-08-20:
Also just to add more detail, these commands are sent via COMMAND_LONG, I'm not running a mission.

DonLakeFlyer, 2018-08-20:
YOu can also use the same QGC I'm using if you want: https://github.com/DonLakeFlyer/VHFCollarTracker. Switch the instrument panel to the Tracker page. Enter in the altitude you want to takeoff to. Click Start. Vehicle will takeoff to specified altitude, then start a rotate/delay loop for a full 360 degrees based on the number of divisions you specify. And then RTL when done 360 degrees.

RomanBapst, 2018-08-20:
@DonLakeFlyer I checked the log and verified that the command arrived correctly in the firmware.
From what I see it looks like the yaw setpoint is updated in the position setpoint triplet but the yaw_valid flag is set to false. Could you also share the SITL log so that I can quickly compare the two?
@dagar Any idea, I'm not that much up to date on this navigator stuff?

DonLakeFlyer, 2018-08-20:
Generating a SITL log. But I just realized I am not flying a completely clean master. There is a small hack in there for command routing from QGC->rPi.. It shouldn't have an affect, and the command does seem to get to the right place but let me get a log from a clean master flight. Let me make sure I didn't screw this up msyelf first. Sorry for not realizing this ahead of time.

RomanBapst, 2018-08-20:
@DonLakeFlyer If you want to try a shot in the dark you could add
```
rep->current.yaw_valid = true;
```
just after this line
https://github.com/PX4/Firmware/blob/master/src/modules/navigator/navigator_main.cpp#L341

DonLakeFlyer, 2018-08-20:
This is failing for me in SITL now as well on clean master. So my change was not causing thing. I think I was running an older master when I was SITL testing. I'll try your change and fly SITL.

DonLakeFlyer, 2018-08-20:
```rep->current.yaw_valid = true;```

That makes it work. Is it safe to try flying for real with that? Would seem so?

DonLakeFlyer, 2018-08-20:
From doing a simple search on yaw_valid it may have a free floating value from random memory. Can't find a set for it. So it may just be lucky that it worked for me in SITL before.

LorenzMeier, 2018-08-20:
@dagar @bkueng Can we get a quick fix in for that?

LorenzMeier, 2018-08-20:
@DonLakeFlyer Should be safe, yes.

RomanBapst, 2018-08-20:
@DonLakeFlyer I think it might have to do something with the new flight task interface. Before using the yaw setpoint we check explicitly if it's valid or not. Probably this check was not there in the old version, but it might be something else. I think it's safe but I will follow up with a fix.

DonLakeFlyer, 2018-08-20:
I'm going to got out a fly a current master with this change in it. I'll let you know what happens with real flight.

RomanBapst, 2018-08-20:
@DonLakeFlyer Good luck and good night!

DonLakeFlyer, 2018-08-20:
Big thanks to folks for helping out so fast.

DonLakeFlyer, 2018-08-20:
Back from flying. Worked perfectly. Thanks

LorenzMeier, 2018-08-21:
@DonLakeFlyer Could you send a PR or @RomanBapst did you already put one up?

DonLakeFlyer, 2018-08-21:
I was assuming that ```rep->current.yaw_valid = true;``` was just a hack and that somewhere the code should be also checking that yaw was actually valid. Otherwise it would not be needed at all and should be removed?

RomanBapst, 2018-08-21:
@LorenzMeier @DonLakeFlyer I did not get to it yet. I don't think it's actually a hack since other valid flags are also set just a few lines below that. The only thing I still need to check if there is a problem with uninitialized memory.

RomanBapst, 2018-08-21:
Let me just push it, if others have a better idea they can comment.

RomanBapst, 2018-08-21:
See #10294

DonLakeFlyer, 2019-02-04:
No idea if this has been fixed for not?

closed 2019-02-18 by stale[bot]