PX4/PX4-Autopilot #10279: Automate Landing Gear for Joystick control

mattgorski, 2018-08-20:
https://github.com/PX4/Firmware/issues/7458

I would like to see more documentation on how to automate this when using gamepad/joystick inplace of RC. Would I just load the mixer and also what is the proper way to load this mixer put on microsd in /etc/mixer does this need a config.txt? Thanks in advance.

hamishwillee, 2018-08-21:
The docs on where to put files to customise startup are in http://dev.px4.io/en/advanced/system_startup.html#customizing-the-system-startup

I do not know if Joystick button customisation is supported. Continued that discussion in #7458

mattgorski, 2018-08-21:
Thank you @hamishwillee much appreciated

mattgorski, 2018-08-30:
I assumed so.  I still have yet to find the process of automating the landing gear by altitude and RTL.  I thought the instructions here said that was possible.

https://github.com/PX4/Firmware/issues/7458

hamishwillee, 2018-08-30:
> I assumed so. I still have yet to find the process of automating the landing gear by altitude and RTL. I thought the instructions here said that was possible.

I'm sure it is. Unfortunately reliant on people with hardware to verify.

closed 2019-02-18 by stale[bot]