PX4/PX4-Autopilot #10282: FlightTasks Auto activation failed

dagar, 2018-08-21:
I've noticed this in a couple logs. Any ideas what's going on?

https://logs.px4.io/plot_app?log=f15dc101-491f-47fc-a547-5427060bb1a7
https://logs.px4.io/plot_app?log=573371f3-4d7a-4a2a-b4d2-c7fd0d9aad73

[image]

Why is it only a warning?

Stifael, 2018-08-21:
Yep. I need to add something such that it does not continuously print that failure. It already uses a limited print https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/mc_pos_control_main.cpp#L334-L342, but that is not enough.

The error message itself is not a bug that just recently got introduced  but rather was always present and cause by statemachine itself: 
Commander set navigation state to auto, which itself sets activates the navigator and the auto-part of the mc_pos_control. The mc_pos_control also requires a valid triplet from the navigator, but that triplet has not been computed yet and therefore the FlightTaskAuto fails to exectute. It is basically a race condition that was always present but just ignored

MaEtUgR, 2019-02-25:
I'm reopening because I saw the race condition is still present in latest master. Even though it was always there and there is a timeout to wait for that case now we should find a better solution. I think reacting with the switch when the data is ready should be our goal instead of switching first and then wait for data that is not yet available. Probably brings us back to static member functions of flighttasks for checks before the task is even running.

not-vibhu, 2019-12-17:
Do we have any fix for this? I'm facing the same issue. Also discussed here https://discuss.px4.io/t/not-able-to-send-attitude-setpoint-commands-in-sitl/7915

ThomasRigi, 2020-02-05:
I experienced the same today, but in addition I also had a "Previous flight task failed, switching to mode 2" Warning just before. Is this linked or did something else cause the additional error?
[image]
Log is here: [https://logs.px4.io/plot_app?log=993534ce-80b3-4a24-9159-358e42c71a51](https://logs.px4.io/plot_app?log=993534ce-80b3-4a24-9159-358e42c71a51)

bkueng, 2020-02-05:
@ThomasRigi it is likely related. What is the closest commit to upstream of that log? And do you have the mission file?

ThomasRigi, 2020-02-05:
@bkueng My modifications are based on v1.10.0 (https://github.com/PX4/Firmware/commit/4f6faac2c86daa14417b33bfbf9c4e44bf17492a). I was testing some stuff for a new hybrid mode for VTOL. I have touched the mc_pos_control files, but only minor modifications that do nothing when it is not in the new hybrid mode (which is not the case here).

Here is the mission file: 
[hybrid square.zip](https://github.com/PX4/Firmware/files/4160299/hybrid.square.zip)
(the way the waypoints are set is also adapted to the new hybrid file, but only standard commands are used)

After a reboot (unplugging and replugging the battery) the mission was executed successfully.

Thanks for looking into it!

bkueng, 2020-02-06:
Thanks @ThomasRigi. What I see is:
- navigator does not send a triplet initially after switching to mission for about 1s
- as a consequence, the position controller outputs these warnings and wants to go into failsafe.
- looking at the perf counters, navigator has a very long cycle (2256 ms), which I'm pretty sure it is waiting for the SD card, and explains the above.

Which SD card are you using?

And in general we need to make sure navigator is more robust against file access delays.

ThomasRigi, 2020-02-06:
A SanDisk Ultra 16GB cat 10 SD card.

Also the initial logging dropout points to an SD card problem.

closed 2022-08-15 by bresch