PX4/PX4-Autopilot #10283: how to get snapdragon adsp firmware and linux image?

zhaoxiaowei1013, 2018-08-21:
I have a snapdragon 801 board ，I hope to transplant PX4 for it，and when i run
``` ./px4 mainapp.config```
it's prompt  in snapdragon shell
```
ERROR [muorb] Initialize Error calling the uorb fastrpc initalize method..
ERROR [muorb] ERROR: FastRpcWrapper Not Initialized
```
and  prompt  in mini-dm
```
[08500/03]  43:45.055  HAP:8253:undefined PLT symbol _Stof (797) /libpx4.so  0303  symbol.c
[08500/03]  43:45.081  HAP:8253:undefined PLT symbol _Stof (797) /libpx4.so  0303  symbol.c
[08500/03]  43:45.105  HAP:8253:undefined PLT symbol _Stof (797) /libpx4.so  0303  symbol.c
[08500/03]  43:45.135  HAP:8252:undefined PLT symbol _Stof (797) /libpx4.so  0303  symbol.c
[08500/03]  43:45.361  HAP:4155:undefined PLT symbol _Stof (797) /libpx4.so  0303  symbol.c
[08500/03]  43:46.141  HAP:8253:undefined PLT symbol _Stof (797) /libpx4.so  0303  symbol.c

```
it maybe mean to update adsp firmware and linux image, but my snapdragon bord does't buy in intrisync, so  i don't have a snapdragon flight SN to regesiter in intrisync to download the flight_3.1.1a_qcom_flight_controller_hexagon_sdk_add_on.zip.

how to do with this things?

mhkabir, 2018-08-21:
This is not an upstream issue which we can fix. Please contact the OEM for support.

closed 2018-08-21 by mhkabir

zhaoxiaowei1013, 2018-08-21:
@ChristophTobler can you  give me some suggest?