PX4/PX4-Autopilot #10284: Feature: RTK heading integration into EKF

AuterionWrikeBot, 2018-08-21:
┆Attachments: <a href="https:&#x2F;&#x2F;storage.www.wrike.com&#x2F;extfile&#x2F;eyJ0IjoiZmExIiwiYWxnIjoiSFMyNTYiLCJjdHkiOiJhcHBsaWNhdGlvblwvanNvbiJ9.eyJhIjoyMDIyMzg2LCJyIjoiVVMiLCJleHAiOjMyNTAzNzA4ODAwLCJpYXQiOjE1MzQ4MzU1MjAsImp0aSI6IjY4MDkzNzg5In0.ohRUUv8K9KmClCirUhKa9LiaAZEXcUBZanErk-dJbus&#x2F;Screen+Shot+2018-08-15+at+9.24.49+pm.png">Screen Shot 2018-08-15 at 9.24.49 pm.png</a> | <a href="https:&#x2F;&#x2F;storage.www.wrike.com&#x2F;extfile&#x2F;eyJ0IjoiZmExIiwiYWxnIjoiSFMyNTYiLCJjdHkiOiJhcHBsaWNhdGlvblwvanNvbiJ9.eyJhIjoyMDIyMzg2LCJyIjoiVVMiLCJleHAiOjMyNTAzNzA4ODAwLCJpYXQiOjE1MzQ4MzU1MjAsImp0aSI6IjY4NTM4MjE3In0.0km-v205pH6hNPcsJLl4UcRvzLAWNwTmCIF94qLcyog&#x2F;Screen+Shot+2018-08-21+at+1.51.34+pm.png">Screen Shot 2018-08-21 at 1.51.34 pm.png</a>

AuterionWrikeBot, 2018-08-21:
➤ Beat Küng commented:

Test logs (the vehicle_gps_position contains a heading field. The range of the heading angle on master is different however):
in-flight:
- manual: https://logs.px4.io/plot_app?log=1b0c9769-5c76-4c85-9947-d7697f723b5c
- PosCtrl+RTK: https://logs.px4.io/plot_app?log=1f35f131-1710-48f6-a01d-1c8409711855
- PosCtrl+no RTK: https://logs.px4.io/plot_app?log=76c3384d-b34b-49d6-85a0-2295c8e4c21f
handheld tests:
https://logs.px4.io/plot_app?log=9214999f-9f62-45bb-87ff-f295edccf88f
https://logs.px4.io/plot_app?log=b62fc6bf-03f0-41a6-8aad-7e93edba8294
https://logs.px4.io/plot_app?log=d9bc48f9-3270-4ff9-8141-c33a006f9f9d
https://logs.px4.io/plot_app?log=ac5404f8-1957-4473-9404-7e1f1473e4a2
https://logs.px4.io/plot_app?log=afd09b37-eb05-489b-a3e2-bf760f62e509

AuterionWrikeBot, 2018-08-21:
➤ Thomas Gubler commented:

Status: heading angle looks ok; need to more clearly define what yaw angle is: is it defined as a projection on the plane or a tait bryan angle (and which convention)? @Paul Riseborough to talk to @Beat Küng

AuterionWrikeBot, 2018-08-21:
➤ Paul Riseborough commented:

Beat, do you know what convention is used to define the heading angle? the angles look OK, but could be either from a 321 or312 Tait Bryan sequence (ie does the angle wrap when pitched past 90 degrees)

AuterionWrikeBot, 2018-08-21:
➤ Paul Riseborough commented:

We could run a test to confirm this

AuterionWrikeBot, 2018-08-21:
➤ Beat Küng commented:

I don't know. I followed the reference manual (http://trl.trimble.com/docushare/dsweb/Get/Document-867971/MB-Two_RM_C_en_Feb2_2018%20(1).pdf), page 62. I did not see anything here, maybe you do.

AuterionWrikeBot, 2018-08-21:
➤ Paul Riseborough commented:

@Beat Küng Can you get some data where you slowly pitch the copter through 360 degrees and then roll through 360 degrees? Observing the yaw output will then tell me how the angle has been defined.

AuterionWrikeBot, 2018-08-21:
➤ Beat Küng commented:

I did the test, this is the log: https://logs.px4.io/plot_app?log=25380997-1d97-4e3c-8006-58fd4d4e52c5

AuterionWrikeBot, 2018-08-21:
➤ Paul Riseborough commented:

Adding yaw to the data structure used by the GPs data buffer in the EKF.

AuterionWrikeBot, 2018-08-21:
➤ Paul Riseborough commented:

Extending exisiting control parameters to enable activation of GPS yaw fusion
Modifying the heading fusion to alwasy use 321 Tait-Bryan definition for GPS yaw and to reject yaw measurement if excessively pitched.

AuterionWrikeBot, 2018-08-21:
➤ Paul Riseborough commented:

@Beat Küng , can you remember if the antennas on the vehicle you tested were fonrt/back or left/right?

AuterionWrikeBot, 2018-08-21:
➤ Beat Küng commented:

They are left-right, with the right antenna being the main one - I also set the GPS_YAW_OFFSET to 90 to account for that.

AuterionWrikeBot, 2018-08-21:
➤ Paul Riseborough commented:

OK, then that means that whether the antennas are left/right or front/back determines if the yaw angle flips when roll or pitch goes thrugh 90 deg. Becasue we don't know that information, I will apply a tilt limit check and only use the GPs yaw when the vehicle is tilted less then TBD deg.

AuterionWrikeBot, 2018-08-21:
➤ Paul Riseborough commented:

The attached figure shows how the yaw from the GPS flips halfway through the first roll rotation and doesn't recover until almost 10 seconds after the rotation finishes. If it is going to a NaN value during that period, then that is OK, but if not then we will need to add protection for that error case.

AuterionWrikeBot, 2018-08-21:
➤ Paul Riseborough commented:

@Beat Küng are you able to do another test where you pitch the vehicle nose up by approximately 45 degrees and then roll the vehicle about +-30deg about the roll axis? That will show if the EKF needs to account for the value of GPS_YAW_OFFSET when it uses the GPS yaw.

AuterionWrikeBot, 2018-08-21:
➤ Beat Küng commented:

I can do that. Do you prefer the heading to be set to NaN if there is no update? Currently the gps message always contains the last value we received from the device.

AuterionWrikeBot, 2018-08-21:
➤ Paul Riseborough commented:

Definitely a NaN, that is what the message definition documentation states.

AuterionWrikeBot, 2018-08-21:
➤ Paul Riseborough commented:

I have the prototype ecl and Firmware changes here: https://github.com/PX4/ecl/tree/pr-ekfGpsYaw and here: https://github.com/PX4/Firmware/tree/ekfGpsYaw-wip

AuterionWrikeBot, 2018-08-21:
➤ Paul Riseborough commented:

What we are missing on the vehicle_gps_position message is an error estimate for the heading message. To work around this I am currently using the yaw observation noise set by the EKF2_HEAD_NOISE parameter.

AuterionWrikeBot, 2018-08-21:
➤ Paul Riseborough commented:

I've been looking through the GPS receiver specs and it does specify a heading accuracy vs baseline length, but that would have to be horizontal length. If we knew the baseline length inside the code, then it would be easy to adjust the observation variance accordingly. However given it is such a small number and we already lock out the yaw measurement if the baseline is within 30 deg of vertical, this complexity is unnecessary. Alternatively we could add a parameter to the GPS code to set a yaw accuracy is published on the vehicle_gps_position message. The EKF can then scale that value based on tilt of the baseline if required.

AuterionWrikeBot, 2018-08-21:
➤ Paul Riseborough commented:

The https://github.com/priseborough/Firmware/tree/ekfGpsYaw-wip branch is working with my bench tests when I fake the GPS yaw. To test this branch, you will need to set EKF2_AID_MASK = 129 and EKF2_MAG_TYPE = 5. That will foirce the EKF to wait until GPS yaw is available before performing the yaw alignment. If EKF2_MAG_TYPE = 0, then the yaw will be set using the magnetomer intially and then reset to the GPS yaw when it becomes available.

priseborough, 2018-08-23:
GPS driver needs to be fixed so that invalid or repeated data is set to NAN. This is causing problems when the vehicle has a yaw rate because the heading measurement updates at a lower rate and the driver is holding the last value.

priseborough, 2018-08-27:
@bkueng I was looking at the GPS driver code today and it looks like if we set the _report_gps_pos.heading class variable to a NAN after each publish event, then that could solve the problem with the heading being updated at a slower rate. See https://github.com/priseborough/Firmware/commit/755bc577f36c639970a8c329cc5d599e8f2e9d20 for the idea I had.

bkueng, 2018-08-27:
Yes exactly, and I would have done the same, but I did not have time to test it yet.

LorenzMeier, 2018-08-27:
@priseborough Can you send a PR with that change? Thanks!

AuterionWrikeBot, 2018-08-28:
➤ Thomas Gubler commented:

PR from Paul for driver here: https://github.com/PX4/Firmware/pull/10346

AuterionWrikeBot, 2018-09-18:
➤ Wrike Bot commented:

@Paul Riseborough All subtasks have been completed or deferred, you might want to complete this task as well.

priseborough, 2018-09-19:
Can't complete yet. The required PX4/ecl PR 
https://github.com/PX4/ecl/pull/49 7has been approved and merged. However the PX4/Firmware PR https://github.com/PX4/Firmware/pull/10355 has run into the recurring problem with insufficient flash memory for the px4fmu-v2 target and is awaiting merging of a couple of PR's that have been submitted to reclaim some flash space.

closed 2018-10-23 by AuterionWrikeBot

thomasgubler, 2018-10-23:
@hamishwillee this one needs documentation as well (@priseborough knows the details)

hamishwillee, 2018-10-24:
@thomasgubler Sure. I have created https://github.com/PX4/Devguide/issues/636 to find out from @priseborough what needs to be documented.