PX4/PX4-Autopilot #10286: PX4 cross referencer

hamishwillee, 2018-08-21:
Can we have a code cross reference/index tool like https://dxr.mozilla.org/mozilla-central/source/ for PX4 code (and QGC etc.)?

@dagar Just had a discussion with @bkueng about this. Essentially we see this as quite easy to set up and a "clearly better than github" way to discover, navigate and link to code.

hamishwillee, 2018-08-21:
FYI We also had a chat about the doxygen output in this context. 

Doxgyen has "useful aspects" (in particular the generated class hierarchy images) but the sidebar is a complete mess. Beat is concerned that it will be a lot of work to get this to be useful, and that effort is not worth the end gain. I share that concern - and if we make it visible there may be the expectation that we will invest the time to get this right (even if we add a disclaimer). So for now we're happy to have this hosted but we won't link it into the other docs yet.

The problems above do not apply to a cross reference tool - it just reflects code structure. Maintenance is therefore low, for clear benefit.

dagar, 2018-08-22:
DXR could be interesting if the code search actually works well.

dagar, 2018-08-22:
To be honest I've never really found doxygen to be that compelling in any project. Maybe I haven't stumbled across the good examples.

One minor benefit is that we could actually use Doxygen to enforce parameter documentation accuracy. I've developed a (bad?) habit of mostly ignoring our comments and documentation strings because they're frequently inaccurate or incomplete.

hamishwillee, 2018-08-25:
@dagar, I think it works well as a tool for generating API docs - as we do for Dronecode SDK. My main concern is that just like discussions of coding standards, doc standards can devolve into an endless round of navel gazing. 
So lets work forward with Doxygen, but not necessarily highest priority.

DXR appears to be quite a good search tool, and the cross linking is excellent. Do you think you might be able to throw up a test site in the next couple of months?

closed 2019-02-18 by stale[bot]