PX4/PX4-Autopilot #10296: Random yawing when RC loss

dakejahl, 2018-08-21:
**Describe the bug**
The vehicle loses RC connection and yaws to a random heading. (We have loiter as our RC loss behavior: NAV_RCL_ACT = 1)
[image]

**To Reproduce**
Current `master` just cause an RC loss. 

**Expected behavior**
Keep the current heading.

**Log Files and Screenshots**
https://review.px4.io/plot_app?log=d9f20475-0558-45f8-9fc9-56d7389f173c

**Additional context**
We are hitting a failsafe which we should not be hitting. `FlightTasks::update()` is failing at line `607` in `mc_pos_control_main.cpp`. This triggers `failsafe()`. Note the vehicle does not actually descend....

[image]

dakejahl, 2018-08-21:
I cannot reproduce this in the simulator.

dakejahl, 2018-08-22:
This only happens when we hit the `failsafe()`

Stifael, 2018-08-22:
it happens because the mc_pos_control_main knows that there is no RC at the same time as commander does, but it takes some time until the commander tells mc_pos_control to switch to auto-loiter.
I will add an additional task that does not require RC. This should fix it

Stifael, 2018-08-22:
> Keep the current heading.

If MPC_YAW_MODE is set to home, then once there is an RC loss the vehicle will enter loiter with heading towards home.

dakejahl, 2018-08-22:
I am not setting MPC_YAW_MODE, what does "towards waypoint" mean for the case of RC loss (loiter)? 

I just flew this PR and I am no longer seeing the failsafe but I am still seeing the yawing (not every time). Notice the first disconnected (postion to loiter) the yaw does not change, but on the second disconnect it does.
[image]
https://review.px4.io/plot_app?log=41843025-0f54-435b-b1a0-567d971bb9af

dakejahl, 2018-08-22:
It is not actually always pointing towards `home`, it sometimes goes to home, sometimes goes to some arbitrary heading like seen in this log.

Stifael, 2018-08-22:
Ok. i will update you tomorrow once I found the reason for that random yawing.

Stifael, 2018-08-23:
I think I found the issue. At least i think it is the same issue that you encountered. 

To reproduce the issue (possible in SITL).
- Fly in Auto Hold
- Do a GoTo command such that the vehicle needs to rotate towards the new target
- Before the vehicle reaches the desire heading (the vehicle wants to point towards the GoTo location) switch into Position. Without any stick inputs, the vehicle should just however.
- unplug the joystick. The vehicle will stay at the current location, but will rotate towards the old GoTo location. 

The cause can be best explained with the prints below:
[image]

From the top to the bottom:
- vehicle is in `Hold` mode -> this is just Auto
- `target` and `position` in xy happens to be [-6.137, 33.6149] and position is [-2.832,-22.560]. In order to point towards the target, the vehicle needs to rotate (just believe me that this is the case). `distance` is just the distance between `target` and `position`.
- at `--------reset triplets` I switched from `Hold` to `Position`. It that happens, the navigator sets the triplets to invalid, which should be some sort of reset (https://github.com/PX4/Firmware/blob/master/src/modules/navigator/navigator_main.cpp#L725-L736). 
- I then unplugged the joystick to cause RC-link loss. In yellow color the commander correctly detects RC-loss. So far everything is as expected.
- due to the RC-loss, the vehicle will correctly enter `Auto`. However, FlightTaskAuto receives for a few iteration a `target` of `0` and then the old `target` followed by another reset of the triplets (not sure why) instead of a target that corresponds to the current location. However, this reset has not caused anything and the `target` is again the old `target`. Finally, there is another reset and this time the `target` will be the current location (not shown)

The time it takes from RC-loss to the correct `target` location is quite short and therefore it is not visible by eye that the `target` is wrong for some very short time period. The yaw, however, is handled slightly different:
https://github.com/PX4/Firmware/blob/1a1a04c0948752dc643c9a466507fe902e18cd4f/src/lib/FlightTasks/tasks/Auto/FlightTaskAuto.cpp#L258-L265
and compute heading method:
https://github.com/PX4/Firmware/blob/1a1a04c0948752dc643c9a466507fe902e18cd4f/src/lib/FlightTasks/tasks/Auto/FlightTaskAuto.cpp#L485-L499

`v` is the distance in xy between the `target` and the current `position` of the vehicle. If this is larger than zero and not smaller than `NAC_ACC_RAD` ( = 2.5 in your setup), then the vehicle will set the `yaw_setpoint` such that the vehicle points towards the `target`. In the example above, this means that during that short time period when it receives the old `target`, the `yaw_setpoint` is set to the heading towards the old `target`. Once the correct `target` is received (= current position), the `v.length()` is close to 0 and for sure smaller than `NAV_ACC_RAD`. From the code snippet above, this means that `yaw_setpoint` will no longer be adjusted and remains whatever it was. Consequently, In our case the `yaw_setpoint` remains pointing towards the old `target`.

In order to fix the actual issue, one needs to dig into the navigator. For the time being, I added now a `yaw-lock`, that becomes active once the vehicle is within `NAV_ACC_RAD`. During the lock, the heading is set to the current yaw.

dakejahl, 2018-08-23:
Thanks! I just flight tested this on a pixracer (6 flights total of ~23 mins flight time) and everything looks good!

closed 2018-09-20 by dakejahl