PX4/PX4-Autopilot #10301: Lightware SF11C LIDAR driver not started on Pixhawk 2.1

klecki98, 2018-08-22:
**Pixhawk doesn't see the LIDAR over i2c port.** 

After connecting the lidar to computer I get the readings but pixhawk doesn't seem to see the lidar. No matter what firmware version I am running (tested on master, stable and beta releases), after running command "sf1xx info" I get information that driver is not started. Command "sf1xx start" results in error "driver failed to start". sf1xx parameter is enabled and set to sf11c. I double-checked and resoldered connection so probably it is not faulty. I attempted to make it work by applying changes mentioned in this [issue](https://github.com/PX4/Firmware/issues/7696). Yet, no change in behavior.

It would be nice if Pixhawk recognized the lidar and got the reading from it.

Kind regards

P.s. You are doing amazing job developing Pixhawk!

fi92, 2018-10-23:
Hi!

My lidar is LW20/B. It is not working with default settings with Pixhawk 2.1 I2C2 because of two reasons:

+ Lightware sensors need to be configured properly at first, according to Rob Gowans from LightWare (https://discuss.ardupilot.org/t/lw20-c-laser-altimeter/31450/33). You can do it with use of LightWareUpgrader software:
      + Connect with use of UART TTL - USB adapter to lidar.
      + Connect with it with use of mentioned software.
      + Update Firmware.
      + Set 'Startup mode' to 'Wait for interface'
      + Set 'Pixhawk I2C compatibility mode' to 'On'
      + Exit

+ PX4 Firmware in version 1.8.1 (the tested one) to run lidar uses distance sensor sf1xx driver, which looks for the device on PX4_I2C_BUS_EXPANSION, while you are probably connecting it to I2C2. You need to replace PX4_I2C_BUS_EXPANSION in the source code (src/drivers/distance_sensor/sf1xx/sf1xx.cpp) with PX4_I2C_BUS_ONBOARD. Rebuild the firmware, upload it and change SENS_EN_SF1XX parameter properly

After reboot you should be able to see, for example with use of QGroundControl's 'Analyze' tool, the current distance given by your distance sensor.

julianoes, 2019-07-09:
@mcsauder is this something you know about?

mcsauder, 2019-07-09:
Hi @klecki98 , if you are still having this issue, ensure that `SENS_EN_SF1XX` is set to `1`, (default is 0), and make sure to follow the setup advice that @fi92 gave above. It's probably also worth double checking that your SDA and SCL lines are connected correctly.

The driver startup in `rc.sensors` is starting the driver on all busses, so it should start whether it's connected to I2C A or I2C B.  I don't have this sensor to test with, but I see no issues with the driver from a cursory read-through.

Let us know if this is still an issue for you!

closed 2019-10-08 by julianoes