PX4/PX4-Autopilot #10304: No valid local position and velocity with `OPTICAL_FLOW_RAD` messages

okalachev, 2018-08-22:
I'm using 1.8.0 version with EKF2, SITL.

Publish faked optical flow messages with zero `integrated_x` and `integrated_y`. The vehicle os not obtaining valid local position nor velocity (`vehicle_local_position.xy_valid`, `vehicle_local_position.v_xy_valid`). This happens when staying still, and in air also.

https://review.px4.io/plot_app?log=1e6bf123-38ed-409d-843d-07ef0d331a7a

(Also, there is some issues with `time_usec` decoding, so I temporarily changed `optical_flow` message's timestamp to `hrt_absolute_time()`.)

okalachev, 2018-08-23:
I found two problems. First, if there zero value in `max_flow_rate` of `optical_flow` message, the local position and velocity are not getting valid.

So something like `f.max_flow_rate = 2.5f` should be added to `handle_message_optical_flow_rad`.

Seconds, the local position and velocity are not getting valid, if `integrated_xgyro`, `integratedyxgyro`, `integrated_zgyro` are `NaN`s, although in that case PX4 should [get](https://github.com/PX4/ecl/blob/master/EKF/optflow_fusion.cpp#L553-L580) gyro data from the PixHawk's gyro.

@dagar , @mhkabir , @priseborough .

mhkabir, 2018-08-23:
Yes, I actually corrected all those three issues in my local branch, however it didn't make much of a difference. I did get valid local position for a while, and then it stopped working in flight.

okalachev, 2018-08-23:
Can you please provide a link to this branch?

mhkabir, 2018-08-23:
I will have to extract the relevant commits; Will do so soon.

Meanwhile, here is a log : https://review.px4.io/plot_app?log=4b678d25-1317-4e5e-a9ad-47ae900ff6cc

mhkabir, 2018-08-23:
OK, I think I just figured out my issue. The vehicle's terrain estimate is completely wrong, and therefore the flow data scaling is wrong, leading to data rejection.

[image]

As for why the terrain estimate is invalid, I think it's just my rangefinder : https://github.com/PX4/Firmware/issues/10264