PX4/PX4-Autopilot #10307: ninja: build stopped: subcommand failed. Makefile:148: recipe for target 'posix_sitl_default' failed

MiguelBennu, 2018-08-23:
Hello,

I followed the instructions here:
http://dev.px4.io/en/setup/dev_env_linux.html#jmavsimgazebo-simulation
using:
ubuntu_sim_ros_gazebo.sh

Then, I installed QgroundControl and it was successful

I followed the instructions here:
https://dev.px4.io/en/setup/building_px4.html

and I was able to 
cd Firmware
git tag –l
git checkout -b tags/v1.7.4beta

but when I tried to do my first build: 
(Using the jMAVSim Simulator) 
make posix jmavsim

I get this error:
ninja: Entering directory `/home/miguel/SFAR/Firmware/build/posix_sitl_default'
[109/589] Generating git version header
FAILED: cd /home/miguel/SFAR/Firmware && /usr/bin/python /home/miguel/SFAR/Firmware/src/lib/version/px_update_git_header.py /home/miguel/SFAR/Firmware/build/posix_sitl_default/src/lib/version/build_git_version.h > /home/miguel/SFAR/Firmware/build/posix_sitl_default/src/lib/version/git_header.log
Traceback (most recent call last):
  File "/home/miguel/SFAR/Firmware/src/lib/version/px_update_git_header.py", line 46, in <module>
    git_branch_name=git_branch_name)
UnicodeEncodeError: 'ascii' codec can't encode character u'\u2013' in position 0: ordinal not in range(128)
[109/589] Generating px4_parameters.c,..._parameters.h, px4_parameters_public.h
ninja: build stopped: subcommand failed.
Makefile:148: recipe for target 'posix_sitl_default' failed
make: *** [posix_sitl_default] Error 1

**To Reproduce**
Steps to reproduce the behavior:
1. Drone switched on '...'
2. Uploaded mission ' N/A 
3. Took off N/A
4. See error

**Expected behavior**
Bringing up the  PX4 console:

What could be going wrong?

Thanks beforehand

M

MiguelBennu, 2018-08-24:
Hi I forgot to mention that I am running Ubuntu 16.04

M

MattMgn, 2018-08-27:
Same error here, but on the master branch when trying to build jmavsim

dagar, 2018-08-28:
Could you post the full output from the beginning? I think something is confused with the versioning from git tags at the arbitrary checkout point.

dagar, 2018-08-29:
If you start from a clean state `make distclean` what's the full output?
I'm looking for the block of cmake configure output at the beginning.

MattMgn, 2018-08-29:
dagar,
Here is the log from the beginning after cleaning the build:

```
matt@matt:~/PX4/Firmware$ make posix jmavsim
-- PX4 VERSION: –l
-- CONFIG: posix_sitl_default
-- Build Type: RelWithDebInfo
-- The CXX compiler identification is GNU 5.4.0
-- The C compiler identification is GNU 5.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Check for working CXX compiler using: Ninja
-- Check for working CXX compiler using: Ninja -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler using: Ninja
-- Check for working C compiler using: Ninja -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
-- Found PythonInterp: /usr/bin/python (found version "2.7.12") 
-- Found PY_jinja2: /usr/lib/python2.7/dist-packages/jinja2  
-- C compiler: cc (Ubuntu 5.4.0-6ubuntu1~16.04.10) 5.4.0 20160609
-- C++ compiler: c++ (Ubuntu 5.4.0-6ubuntu1~16.04.10) 5.4.0 20160609
Submodule 'msg/tools/gencpp' (https://github.com/ros/gencpp.git) registered for path 'tools/gencpp'
Submodule path 'tools/gencpp': checked out 'b41ee3060badd660662e21cd9d4f81971c87a420'
Submodule 'msg/tools/genmsg' (https://github.com/ros/genmsg.git) registered for path 'tools/genmsg'
Submodule path 'tools/genmsg': checked out 'a189fc78558e7276df59d2961cfe4f8b4de08a8b'
Submodule 'src/lib/DriverFramework' (https://github.com/PX4/DriverFramework.git) registered for path 'src/lib/DriverFramework'
Submodule path 'src/lib/DriverFramework': checked out '3bdfdf3db201d19dc243a1a1f1f3f85b55d08212'
Submodule path 'src/lib/DriverFramework/dspal': checked out '9b46b4a57f230672ee0806a523963af70bc44f1c'
Submodule path 'src/lib/DriverFramework/dspal/cmake_hexagon': checked out '07168bd5715818802b78f674816ec851307998a7'
Submodule 'src/lib/ecl' (https://github.com/PX4/ecl.git) registered for path 'ecl'
Submodule path 'ecl': checked out '56b8bb08a16f0e86f7c7dfe458f0d2d3082e92e4'
Submodule 'src/lib/matrix' (https://github.com/PX4/Matrix.git) registered for path 'matrix'
Submodule path 'matrix': checked out 'b815fc97c4e686a93a8074f27d1830a031b0d38d'
-- PX4 ECL: Very lightweight Estimation & Control Library v0.9.0-583-g56b8bb0
Submodule 'src/drivers/gps/devices' (https://github.com/PX4/GpsDrivers.git) registered for path 'devices'
Submodule path 'devices': checked out 'aef48b04ff9b95350c4e5e507eea72a9e971d6a4'
Submodule 'mavlink/include/mavlink/v2.0' (https://github.com/mavlink/c_library_v2.git) registered for path '../../../mavlink/include/mavlink/v2.0'
Submodule path '../../../mavlink/include/mavlink/v2.0': checked out 'fe2349dea9a88fb5b873eff2a59a88125943fd61'
Submodule 'Tools/sitl_gazebo' (https://github.com/PX4/sitl_gazebo.git) registered for path '../../Tools/sitl_gazebo'
Submodule path '../../Tools/sitl_gazebo': checked out '821c55ca714d7acf93d8ed96af71237b8437e488'
Submodule path '../../Tools/sitl_gazebo/external/OpticalFlow': checked out '549cc7cacb01bb71b2386586305c889d3651cfcc'
Submodule path '../../Tools/sitl_gazebo/external/OpticalFlow/external/klt_feature_tracker': checked out 'a0d242294ef638d8fa422ab43f0d476ba37a15a6'
Submodule 'Tools/jMAVSim' (https://github.com/PX4/jMAVSim.git) registered for path '../../Tools/jMAVSim'
Submodule path '../../Tools/jMAVSim': checked out '136e92eb8a0131b63c165d5f32dd74c3a2b28993'
Submodule path '../../Tools/jMAVSim/jMAVlib': checked out 'afa488d166813038c28572907eacf6eaa54f3c7c'
-- Configuring done
-- Generating done
-- Build files have been written to: /home/matt/PX4/Firmware/build/posix_sitl_default
ninja: Entering directory `/home/matt/PX4/Firmware/build/posix_sitl_default'
[27/672] Generating git version header
FAILED: cd /home/matt/PX4/Firmware && /usr/bin/python /home/matt/PX4/Firmware/src/lib/version/px_update_git_header.py /home/matt/PX4/Firmware/build/posix_sitl_default/src/lib/version/build_git_version.h > /home/matt/PX4/Firmware/build/posix_sitl_default/src/lib/version/git_header.log
Traceback (most recent call last):
  File "/home/matt/PX4/Firmware/src/lib/version/px_update_git_header.py", line 46, in <module>
    git_branch_name=git_branch_name)
UnicodeEncodeError: 'ascii' codec can't encode character u'\u2013' in position 0: ordinal not in range(128)
[27/672] Generating parameters.xml
ninja: build stopped: subcommand failed.
Makefile:148: recipe for target 'posix_sitl_default' failed
make: *** [posix_sitl_default] Error 1
```

matt

dagar, 2018-08-29:
Thanks, that's what I was looking for.

The build is failing because you don't have the tags in your clone (`PX4 VERSION: –l`). The quick fix is `git fetch --all --tags`, but we could probably update it to fail gracefully as well.

MattMgn, 2018-08-29:
Even after fetching the tags I still have this error, although `git tags -l` returns a full list of tags.
I am not familiar with tags, but if I checkout on another branch (v1.8.0-rc0 for example) i am able to build

dagar, 2018-08-29:
What if you fetch everything (including tags), distclean, and get back on master?

``` Bash
    git fetch --all --tags
    make distclean
    git checkout master
    git reset --hard origin/master

    make posix_sitl_default
```

MattMgn, 2018-08-29:
The tag is still not configured  ` -- PX4 VERSION: –l`

dagar, 2018-08-29:
Oh hold on, I think the problem is that you accidentally created a tag with a name that's choking the version scripts. 

Try `git tag -d –l`, then rebuild.

If that doesn't work do a fresh clone on the side.

MattMgn, 2018-08-29:
Yes, you are right! Actually, the wiki proposes:
```
 git tag –l
```
But the right syntax should be:
```
 git tag -l
```
This is quite confusing :s
Thank you for your help

matt

dagar, 2018-08-29:
Let's get that fixed in the wiki. 
https://github.com/PX4/Devguide/pull/592

closed 2018-08-29 by dagar

MiguelBennu, 2018-08-29:
Thank dagar and MattMgn