PX4/PX4-Autopilot #10315: NXPhlite: hardfault when both USB and telemetry are connected

Nielesonimusso, 2018-08-24:
When both USB and Telemetry radio (through UART4) are connected to the NXPhlite, it hardfaults at random moments. Attached is a log that occurs on the debug console whenever the hardfault occurs.
[Hardfault USB AND UART4.txt](https://github.com/PX4/Firmware/files/2317757/Hardfault.USB.AND.UART4.txt)

Nielesonimusso, 2018-08-24:
Update: hardfault does not seem to occur when RC provided as PPM, but it does occur when RC is provided as SBUS.

Nielesonimusso, 2018-08-24:
Update 2: hardfault also just occured when only SBUS and USB were connected, issue seems to be related to SBUS, not USB/UART4

Nielesonimusso, 2018-08-24:
Update 3: after some more hardfaults, it seems that they are unrelated to either SBUS, UART4 or USB: I have seen hardfaults happening with arbitrary combinations of the three connected. Will report when I have found a pattern

davids5, 2018-08-24:
@Nielesonimusso - Thank you! This is super helpful. If you can capture the log from boot including the fault. The will help too. You are on the correct path, find and document the minimum test setup to replicate. Even upload the parameters. Describe the version of QCG. What is connected etc.

Nielesonimusso, 2018-08-24:
[full_log_crash.txt (includes log from start-up and log from reboot start-up)](https://github.com/PX4/Firmware/files/2318881/full_log_crash.txt)

[crash_params.txt](https://github.com/PX4/Firmware/files/2318887/crash_params.txt)

QGC version is Daily Build `c7a171eb0 2018-08-21 20:40:28 -0400`

during the hardfault, connected components were:
- power supply (unconnected; power from USB)
- USB
- PPM RC (through frsky connector)
- arming switch
- GPS module (Ublox M8N)
- Telemetry radio (through UART4)
- Debugger (DCD-LZ connector, only using the console)

no ESCs were connected

I won't be able to test more until coming Monday, unfortunately

Nielesonimusso, 2018-08-24:
[full_log_crash_2.txt](https://github.com/PX4/Firmware/files/2318903/full_log_crash_2.txt)

another full log resulting from a crash that happened just now. Note boot after the crash does not complete, this also happens sometimes. The up_stackdump also looks different for this one, hence why I am uploading this one as well.

Setup for this crash is exactly the same as for the previous one

Nielesonimusso, 2018-08-27:
Issue persist under latest master (f7a20ac).  Attached more logs starting from first hard fault. Setup is the same as before, only with RC on SBUS for first crashes, and RC disconnected before the last crash.

[full_log_crash_3.txt](https://github.com/PX4/Firmware/files/2323065/full_log_crash_3.txt)

davids5, 2018-08-27:
@Nielesonimusso - This is proving to be a hard to tack down bug. I will keep trying.

Nielesonimusso, 2018-08-28:
are there any tests that I can perform to help? I will only be able to perform tests on upcoming Friday.

davids5, 2018-08-28:
@Nielesonimusso - Thank you for offering. Anything to reduce the number of variables would surely help. The issue I am having is none of that changes I have made change the problem any any way and my normal method to debug this is not working.

Nielesonimusso, 2018-08-31:
I did some more tests, connecting one component at a time and keeping the NXPhlite powered for 15 minutes. The components I used are 

- telemetry radio
- RC with PPM signal
- RC with SBUS signal
- GPS

I did two tests for each: one with power over USB and one with power from batteries. 

No crash occured whatsoever...

I also did tests on battery power making combinations of two of the components (RC only on SBUS), and this also did not result in any crash

There are obviously more tests that can be done (more components/combinations, longer time, ...), but this is what I got so far. I hope it is useful

Nielesonimusso, 2018-09-05:
Have done some more tests today, have had a lot of crashes again, most notable is that all crashes occur whenever the USB cable is connected; even when only USB is connected (and nothing else), a crash occurred. All tests of today I have run for 15 minutes, all crashes occurred within the first 10 minutes. I am now running a longer test with everything _but_ USB connected, and so far I have seen no crashes.

My conclusion for now based on the latest tests is therefore that the issue seems to come from the USB being connected. This is a bit inconsistent with my previous results (no crashes at all, so also no crashes when USB was connected), but this could have to do with the duration of tests.

Note that I have QGroundControl running at all times, connected to the NXPhlite either through USB or through the telemetry radio (UART4). I also have the debug console connected at all times, to see crash logs (I have not saved the logs, though there seemed to be nothing different from the previous crash logs I have put on this thread).

davids5, 2018-09-05:
@Nielesonimusso Thank you for continuing to test. I think you and I are seeing similar results.

 When you say nothing else connected do you mean that you've removed all the cables from the Drone except for the USB?

If not that would be a good test. 

What I'm seeing on my side is corruption on the stack. The difficulty is finding what is corrupting it.

Nielesonimusso, 2018-09-05:
Only USB connected means all other cables are disconnected, yes.

jarivanewijk, 2018-09-12:
@Nielesonimusso and I have done tests again with the latest master firmware (commit 73fe4ec). We still get the hardfaults, I have included the log.

To us it seems like the issue only occurs when we have USB connected and have QGroundControl communicating with the FMU. When QGroundControl is not running, no hardfaults occur. Without QGroundControl, we have had it running for a combined duration of at least an hour without any hardfaults.

Other devices and connetions don't seem to matter.

[usb_hardfault.txt](https://github.com/PX4/Firmware/files/2375819/usb_hardfault.txt)

davids5, 2018-09-13:
@jarivanewijk and @Nielesonimusso - I concur with the findings that the hard fault only happens over USB while QGC is running. - Thank you both for narrowing it down.

We are working on getting the list of open NXPhlite items closed. This one will be last, as it has been very difficult to find the root cause. My plan is to get the USB monitor on it and see if the stack overwrite is based on consistent USB events. If that proves to be fruitless. I will add ETM trace and see if it is CPU is the cause as opposed to DMA.

davids5, 2018-09-21:
Closed via https://github.com/PX4/Firmware/commit/d2faf5f31d2fbfc4466a5e2c885c2fd445b9861d

closed 2018-09-21 by davids5