PX4/PX4-Autopilot #10318: NXPhlite: No PWM to ESCs

Nielesonimusso, 2018-08-24:
After a full setup, when arming the NXPhlite, the output PWM towards the ESCs stays at 900 (disarmed). Even when moving the throttle up and down, there is no change in PWM. It concerns a generic quad-copter X configuration, attached an export of all the parameters after the complete setup. 

[params_full_setup.txt](https://github.com/PX4/Firmware/files/2318827/params_full_setup.txt)

Using the `pwm test` command in Nuttshell also does not seem to affect any of the PWM outputs.

Note that USB was connected to the NXPhlite while performing most of these tests (all but one of the testing NXPhlites do not have a functioning Telemetry (no U50 rework)), but when that was the case, the `CBRK_USB_CHECK` was set to 197848 (disable USB check).

igalloway, 2018-09-10:
Both of the sequences below should spin all four motors.
For me - The first one seems to partially work , and spins all but the first motor
The second `pwm test` method will also not spin the first motor. sometimes one or more of the other motors will spin. This is also not consistent on the same unit from test to test.
- Because there are inconsistencies, we have some suspicion of the ESCs, as they beep and initialize at the `pwm_arm` command. These are the Opto-isolated 40A escs that come with the HGDrone.
- I'm wondering if they are picking up some pwm "setup code" incorrectly
- Will test with another set of ESCs
- Will put a scope on the pwms 

```
nsh> fmu stop
nsh> fmu mode_pwm6
nsh> fmu test
```

Below:
-p is the pwm value
-c is the channel 
```
fmu stop 
fmu mode_pwm6 
pwm arm
pwm test -p 1200 -c 1234  
```

jarivanewijk, 2018-09-10:
@igalloway I checked with a scope, both commands seem to work fine here. With the first command the PWM signals on all 4 channels go up and down nicely, and with the second command they all have the expected constant value.

However, the initial issue described by @Nielesonimusso is still there. When I try to arm the drone with both the safety switch and the controller, the PWM stays at the idle throttle PWM (900us). I can put the throttle up and down, and in the console I see messages about takeoff and landing detected, but the PWM does not go up or down.

lericson, 2018-09-12:
Try this change

```diff
commit de177b41bd6293572020858dec441cb974eab9e4
Author: Ludvig Ericson <ludvig@lericson.se>
Date:   Wed Sep 12 18:14:10 2018 +0200

    Set _safety_off = _safety_btn_off if a hardware button exists

diff --git a/src/drivers/px4fmu/fmu.cpp b/src/drivers/px4fmu/fmu.cpp
index a6927563e..23da99ecd 100644
--- a/src/drivers/px4fmu/fmu.cpp
+++ b/src/drivers/px4fmu/fmu.cpp
@@ -1034,6 +1034,8 @@ PX4FMU::run()
 void
 PX4FMU::cycle()
 {
+	bool updated = false;
+
 	while (true) {
 
 		if (_groups_subscribed != _groups_required) {
@@ -1273,10 +1275,11 @@ PX4FMU::cycle()
 			}
 		}
 
-#endif
+		_safety_off = _safety_btn_off;
+
+#else
 
 		/* check safety button state */
-		bool updated = false;
 		orb_check(_safety_sub, &updated);
 
 		if (updated) {
@@ -1287,6 +1290,8 @@ PX4FMU::cycle()
 			}
 		}
 
+#endif
+
 		/* check arming state */
 		orb_check(_armed_sub, &updated);
 
```

lericson, 2018-09-12:
This looks like the exact same thing as in #10457

davids5, 2018-09-13:
@dagar - is the the cause of what we were debugging?

davids5, 2018-09-13:
Fixed in #10477

closed 2018-09-13 by davids5