PX4/PX4-Autopilot #10328: startup scripting establish common subset (sh, nsh, bash)

dagar, 2018-08-25:
- environment variables
 - conditionals
   - multiple conditions (AND, OR)
 - equality tests
 - airframe autostart (generate rc.autostart?)
 - shebang

We should do a quick pass to figure out what's actually needed, then establish the limited subset. A simple test should be written that runs on all platforms to verify (NuttX, Linux (bash, sh, dash), OSX (older bash?).

dagar, 2018-09-12:
Linux: bash >= 4
NuttX: bash like mostly sh compatible
OSX: bash 3.2

julianoes, 2018-09-13:
I think we should find out what NuttX supports and validate for that.

bartslinger, 2018-09-26:
Bebop 2 runs busybox which apparently has ash shell by default, I don't which version.

dagar, 2019-01-30:
Still needed.