PX4/PX4-Autopilot #10330: EKF2 vision pose aid influence the velocity estimation.

eleboss, 2018-08-26:
**Describe the bug**
I am using VINS-mono VIO as a external vision support, and turn on the EKF2 aid, sending topic to /mavros/vision_pose/pose at 10HZ, when the vision system works fine, the EKF2 give a relative accurate estimation, however, when the external vision system start to drift, the velocity estimation become very inaccurate, and give a impossible velocity, this will directly lead to a crash in velocity control.

Velocity feedback like. from /mavros/local_position/odom
```
header: 
  seq: 391334
  stamp: 
    secs: 1535299548
    nsecs: 866225664
  frame_id: "map"
child_frame_id: "base_link"
pose: 
  pose: 
    position: 
      x: -0.312624961138
      y: -4.13867712021
      z: 8.31071472168
    orientation: 
      x: 0.403164698857
      y: 0.156174051339
      z: 0.0814457080274
      w: -0.89801704281
  covariance: [1e-06, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1e-06, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1e-06, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1e-06, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1e-06, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1e-06]
twist: 
  twist: 
    linear: 
      x: 0.220971111536
      y: -1.39470153404
      z: -0.0521862682266
    angular: 
      x: -0.0481302216649
      y: -0.022168552503
      z: -0.00464850105345
  covariance: [0.0001, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0001, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0001, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0001, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0001, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0001]

```

also, when start to using the vision aid, the drone shacking more frequently, I think this may be introduced by the vision noise, making the total velocity not so accurate.

All in one, is there any way to separate the vision aid and velocity estimation? Let the vision noise not influence the velocity estimation, make the whole system more robust.

**Drone (please complete the following information):**
- Describe the type of drone.
DJI f550
- Photo of the IMU / autopilot setup if possible.
pixhawk4 with 1.8.0 stable release

mhkabir, 2018-08-26:
Please send a log from boot. (Use `SDLOG_MODE` parameter).

However I'm not very sure what we can do if you're vision system is not functioning correctly.

eleboss, 2018-08-27:
@mhkabir Hi!
I log some data from px4, and doing multiple tests.
Here is the log:
1.without vision position feedback(following the speed at beginning then start to swag):
https://logs.px4.io/plot_app?log=393564ec-92a7-47e0-b5a2-cb13cd1a893f
2.with vision feedback:(following the speed well)
https://logs.px4.io/plot_app?log=98cb0a5f-eaa5-4a37-8286-0e819d335974

In the two tests, the control script are the same, set both x and y axis at 0.06m/s
(as the tfmini bug fixed, z position control can be quite stable, so I am not included here)
```
position_pub = rospy.Publisher('/mavros//setpoint_raw/local', PositionTarget, queue_size=1)
    stamp = rospy.get_rostime()

    raw_msg = PositionTarget(coordinate_frame=PositionTarget.FRAME_LOCAL_NED,
                             type_mask=
                                       PositionTarget.IGNORE_PX + PositionTarget.IGNORE_PY +  PositionTarget.IGNORE_PZ +
                                       PositionTarget.IGNORE_AFX + PositionTarget.IGNORE_AFY + PositionTarget.IGNORE_AFZ +
                                       PositionTarget.IGNORE_YAW  + PositionTarget.FORCE,

                             yaw = 0)
    raw_msg.header.stamp = stamp
    output_x = 0.06
    output_y = 0.06
    raw_msg.velocity.x = output_x
    raw_msg.velocity.y = output_y                             
    raw_msg.velocity.z = output_z
    position_pub.publish(raw_msg)
```

Here I optimize the slam system, make it not so easy to drift, since the drift is really a hard problem.
However, I found more interesting things, the velocity feedback looks same in the log files, but the actual performance different, I am trying to take some video demos to fully show this. 

Here I give a velocity setpoint figure, the bottom one, when the vision system is enable, the drone velocity setpoing following the 'velocity' well, well the actual performance is not, it swag.
[image]

eleboss, 2018-08-27:
video is ready now!
1.without vision aid, log and video
[](  https://www.youtube.com/watch?v=frP3VX3a-t4&feature=youtu.be)
[](https://logs.px4.io/plot_app?log=a91a3b09-f01a-4be1-8248-b7a4c45fc815)
2.with vision aid ,log and video
[](https://youtu.be/-Fo0F79yQDA)
[](https://logs.px4.io/plot_app?log=d5b29dc2-40c7-4b66-8507-cd0e83fd2b59)

eleboss, 2018-08-27:
This really confused me, am I wrong about something?

geetesh, 2018-09-01:
I recently did a stable position hover with VINS VIO only as feed. I think feeding both, the position estimate and speed estimate works better than feeding the position alone. Try this and see if it helps you.
Also, try to set the VINS VIO covariances higher, because the in-flight imu-sensor noise due to vibrations will have severe effects compared to hand-held trials.

eleboss, 2018-09-02:
@geetesh 
Hi!
Very appreciate for the useful sugestion！ But I can't find any place to send speed feedback in /mavros/vision_pose/pose topic, Set the IMU noise higher is a very good suggestion, and help my VINS system works much better.

During my fly tests, under a well worked VINS , It seems that Turnning on the **rotate external vision** in **EKF2_AID_MASK**  can make the ekf2 works, if not, I always get the **critical navigation error** . 
Have you meet the same situation ? 
BTW, my VINS estimation is work under ENU in ROS, and rotate to NED, checked in QGC.

closed 2019-02-18 by stale[bot]